OllisGit/OctoPrint-SpoolManager #177: Spool manager doen't show the spools features

davbcr, 2021-05-27:
It happens after updating Octoprint 1.6.1
[image]
[image]

OllisGit, 2021-05-27:
Hi @davbcr ,

do you use the plugin "UI Customizer"?

OllisGit, 2021-05-28:
Hi @davbcr ,

- which version of OP do you use?
- is the problem in all browsers (chrome, firefox, ...)?
- which screenresolution do you use?
- when you change the zoom-level in the browser it the dialog then visible?
- what other plugins do you use?

davbcr, 2021-05-28:
which version of OP do you use? - 1.6.1
is the problem in all browsers (chrome, firefox, ...)? i've only chrome used and installed.
which screenresolution do you use? - it's not working both 1366x768 and 1600x900
when you change the zoom-level in the browser it the dialog then visible? - it doesn't
what other plugins do you use? 
[image]
[image]
[image]
UI customizer install this morning just to see if it changes something,but it doesn't.

STekZeNIO, 2021-05-28:
Confirmed the same issue persists, since 1.6.1

Bootscreen, 2021-05-29:
I have a Quick and Dirty Solution:
in the UI Customizer add this Custom CSS:
```
#dialog_spool_select .modal-footer{
    position: static !important;
}
```

This works for me =)

davbcr, 2021-05-30:
I try and unexplecably it works only at 175% zoom for me....... at least i can insert new spools now......

lightmaster, 2021-06-01:
> #dialog_spool_select .modal-footer{
>     position: static !important;
> }

Can confirm, this Custom CSS works for me too.

gn0m3r4l, 2021-06-02:
Same for me.
Plugin Version 1.4.3
Octoprint 1.6.1
Browser independent error

Unfortunately, I cannot tell when this occurs first time.