OllisGit/OctoPrint-SpoolManager #178: When starting a print, the selected spool from Spool manager shows initial totyal spool weight not currenbt spool weight

clarkiej, 2021-05-27:
Previous versions showed current spool weight when selecting the spool for printing. Now when you select the spool the spool choice shows teh initial spool weight (1000 gms) instead of the current spool weight (854 gms) as it did previously. It makes it difficult to choose spools if you have more than one spool of the same type/colour of differing remains in your spool catalogue.

OllisGit, 2021-05-27:
@clarkiej please attach a screenshot of the selected spool-dialog and also the popup.
Btw. the poup before print, shows the same weight as in the selection-box, rigth?

clarkiej, 2021-05-27:
Correct. It is just when you select the spool it now shows "Grey PLA (1000gm)" when previous it showed "Grey PLA (670gm)". Once selected it shows the remaining filament. I will do a screenshot once my current print finishes. Kieran ClarkeMobile +61418940005

codingcatgirl, 2021-05-28:
I might have the same problem here, but it looks different. There is different data displayed for the same spool. I'm not shown the initial weight of the spool but something is definitely wrong
[image]

- The print i'm trying to start uses 58g of filament so… it doesn't make sense to be related to that.
- I do not have a spool that currently has 276g left, so it can't be mistaking it for a different spool.

In loadSpoolsByQuery, the following data is returned:
```
      "remainingLength": "-256064",
      "remainingLengthPercentage": "",
      "remainingPercentage": "19.2",
      "remainingWeight": "191.8",
```

but then, in the allowedToPrint call, the following data is returned:
```
    "reminderSpoolSelection": [
      {
        "bedOffset": null,
        "enclosureOffset": null,
        "material": "PETG",
        "remainingWeight": 276.3485105427127,
        "spoolName": "Prusament Jet Black 3",
        "toolIndex": 0,
        "toolOffset": null
      }
    ]
```

If i try to edit the spool, the value displayed it 191.8, so i assume that's the correct one… or is it?

The key difference seems to be that loadSpoolsByQuery runs it through transformSpoolModelToDict, and allowedToPrint doesn't. allowedToPrint calculates the remaining weight based on the total and used values from the database, while allowedToPrint just reads the remaining value from the database. That might cause an inconsistency, but it makes me wonder… why is the remaining value not in sync with the value of total weight and used weight? Could there be something wrong with how the multi-tool code commits the usage data to the database?

I don't know what your database is supposed to hold and store precisely, so it's for you to find out @OllisGit :) Hope this helps.

PS: Wondering if this might be ralted to #175 but it probably isn't.

codingcatgirl, 2021-05-28:
Ok, i had toi look some more into it:

The culprit is here:
https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/__init__.py#L377

After the print, only used filament is updated, not remaining filament. And i'm to blame, since this code was introduced by my pull request.

Question: why are used filament and remaining filament stored as seperate values at all. Wouldn't it be a good idea to drop one of them from the database and always calculate it and if the user puts in a new value, just convert it into a new value for used filament instead?

codingcatgirl, 2021-05-28:
Okay, apparently i'm no to blame after all, because before my edits, the previous code also didn't update the remainingWeight field: https://github.com/OllisGit/OctoPrint-SpoolManager/blob/45f730b0df7c29d8cd44e895cdbe143a776990b4/octoprint_SpoolManager/__init__.py#L328 So something else must be going on?

It looks like you just mistakenly added code to read the remainingWeight value from the spool model isntead of calculating it:
https://github.com/OllisGit/OctoPrint-SpoolManager/commit/3dbd15c429d4853f79a7eae2686ea71b6c300cae#diff-4963735b757dd922fea5143ec41a3b92d3464e9d242cbc11f8f243f716bc303aR324

Which is an easy mistake to make, since it seems like an obvious thing to do, but that begs the question even more why that field exists if you are not supposed to read data from it?

clarkiej, 2021-05-28:
Update:

 

This is only happening with a new spool I added AFTER I performed the update:

 

 

With a spool in the system prior to the update, it displays correctly:

 

 <https://user-images.githubusercontent.com/3868507/119974004-78678200-bfb4-11eb-9520-a836ebd9bc39.png>

clarkiej, 2021-05-28:
Update:

This is only happening with a new spool I added AFTER I performed the update:
[image]

  
[image]

With a spool in the system prior to the update, it displays correctly:

  
[image]
[image]

clarkiej, 2021-05-29:
Update.: If the spool weight is left blank, the spool weight does display when you go to start a print, BUT, the spool weight shown is the spool weight remaining after the previous print -n. It has not updated the total remaining after the previous print, but the filament remaining displayed in the spool catalogue is correct:

[image]

Filament remaining in spool catalogue showing correctly.

Start the next print and the filament remaining is the amount that was left at the START of the previous print, not the end:

[image]

clarkiej, 2021-10-29:
> Hi @clarkiej <https://github.com/clarkiej>  ,
> in latest release 1.5.0 I change the behaviour. The dialog should now present the correct values.
>
> Please try and give me a feedback.
> Olli

Olli,

 

The new release is working great.

I especially like the new picklist format. It looks sweet

 

Regards,

 

Kieran Clarke

closed 2021-11-03 by OllisGit