OllisGit/OctoPrint-SpoolManager #179: "Shopping List" print out

OllisGit, 2021-05-27:
Initial ticket, see #141

Option for creating a printable page for:
- Single Spool
- Multi Spool

chemixt, 2021-11-17:
Not sure what happened, but I recently upgraded python from 2 to 3 and restored my backup, I also had to reinstall the GPX plugin (using with a MakerBot 2x).   Now whenever I go to print something I get this message from Spool-Manager:

_Octopi.local says
Attention: Needed filament could not calculated (missing metadata or spool-fields)

There are no spools selected for the following tools despite them being used (maybe) by this print:
- Tool 0
- Tool 1

Do you want to start the print without selected spools?  

Any ideas what I need to do to fix it?

OllisGit, 2022-01-07:
tracked here: #209

closed 2022-01-07 by OllisGit