OllisGit/OctoPrint-SpoolManager #180: SPM:ERROR:DatabaseManager - Could not insert the spool into the database

moorsey, 2021-05-29:
Trying to re-import data after filling in some more fields, but now getting this red pop-up error popping up, even though the plugin shows the successful import of spools

CSV data and copy of data from octoprint.log below

Have tried deleting the database from the "Storage" tab, removing and re-installing the plugin and selecting replace DB when importing

Can't see anything in the logs that indicates a line from the CSV that is potentially tripping things up

CSV:

```
Spool Name,Color Name,Color Code [hex],Vendor,Material,Density [g/cm3],Diameter [mm],Diameter Tolerance[mm],Flow rate compensation [%],Temperature [C],Bed Temperature [C],Enclosure Temperature [C],Total weight [g],Spool weight [g],Used weight [g],Total length [mm],Used length [mm],First use [dd.mm.yyyy hh:mm],Last use [dd.mm.yyyy hh:mm],Purchased from,Purchased on [dd.mm.yyyy],Cost,Cost unit,Note
Eryone-PLA-Blue,Blue,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,392,,,,,Amazon,20.03.2020,18,£,
Eryone-PLA-Yellow,Yellow,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,426,,,,,Amazon,20.03.2020,18,£,
Eryone-PLA-Red,Red,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,831,,,,,Amazon,06.02.2020,18,£,
Eryone-PETG-Black,Black,,Eryone,PETG,1.24,1.75,0.05,,,,,1000,,815,,,,,Amazon,20.03.2020,18,£,
Eryone-PLA-Green,Green,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,591,,,,,Amazon,20.03.2020,18,£,
Ender-PLA-White,White,,Ender,PLA,1.24,1.75,0.05,,,,,500,,95,,,,,Ender,,0,£,
Eryone-PLA-Black,Black,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,963,,,,,Amazon,02.04.2020,18,£,
Eryone-PLA-White,White,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,865,,,,,Amazon,,23,£,
Eryone-PLA-Rainbow,Rainbow,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,516,,,,,Amazon,24.08.2020,20,£,
Eryone-PLA-Carbon Black,Carbon Black,,Eryone,PLA,1.24,1.75,0.05,,,,,1000,,521,,,,,Amazon,01.03.2021,21.59,£,

```

Log file:

```
UnicodeDecodeError: 'utf8' codec can't decode byte 0xa3 in position 0: invalid start byte
2021-05-29 08:28:54,833 - octoprint.plugins.SpoolManager - WARNING - SendToClient: error#DatabaseManager#Could not insert the spool into the database. See OctoPrint.log for details!
2021-05-29 08:28:54,839 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-05-29 08:28:54,840 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
2021-05-29 08:28:54,845 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-05-29 08:28:54,845 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('BEGIN', None)
2021-05-29 08:28:54,850 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('BEGIN', None)
2021-05-29 08:28:54,851 - octoprint.plugins.SpoolManager.DatabaseManager - ERROR - Could not insert Spool into database
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 905, in databaseCallMethode
    spoolModel.save()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 6556, in save
    pk = self.insert(**field_dict).execute()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1907, in inner
    return method(self, database, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1978, in execute
    return self._execute(database)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2745, in _execute
    return super(Insert, self)._execute(database)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2474, in _execute
    cursor = database.execute(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 3156, in execute
    sql, params = ctx.sql(query).query()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 613, in sql
    return obj.__sql__(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2722, in __sql__
    self._simple_insert(ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2576, in _simple_insert
    return self._generate_insert((self._insert,), ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 2694, in _generate_insert
    return ctx.sql(CommaNodeList(all_values))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 613, in sql
    return obj.__sql__(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1773, in __sql__
    ctx.sql(self.nodes[n_nodes - 1])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 613, in sql
    return obj.__sql__(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1771, in __sql__
    ctx.sql(self.nodes[i])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 613, in sql
    return obj.__sql__(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/peewee.py", line 1379, in __sql__
[... 13 lines cut ...]
2021-05-29 08:28:58,246 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
2021-05-29 08:28:58,251 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-05-29 08:28:58,255 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-05-29 08:28:58,257 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
2021-05-29 08:28:58,263 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-05-29 08:28:58,273 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT COUNT(1) FROM (SELECT 1 FROM "spo_spoolmodel" AS "t1") AS "_wrapped"', [])
2021-05-29 08:28:58,297 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."updated", "t1"."version", "t1"."originator", "t1"."isActive", "t1"."isTemplate", "t1"."displayName", "t1"."vendor", "t1"."totalWeight", "t1"."spoolWeight", "t1"."usedWeight", "t1"."remainingWeight", "t1"."totalLength", "t1"."usedLength", "t1"."code", "t1"."firstUse", "t1"."lastUse", "t1"."purchasedFrom", "t1"."purchasedOn", "t1"."cost", "t1"."costUnit", "t1"."labels", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml", "t1"."material", "t1"."materialCharacteristic", "t1"."density", "t1"."diameter", "t1"."diameterTolerance", "t1"."colorName", "t1"."color", "t1"."flowRateCompensation", "t1"."temperature", "t1"."bedTemperature", "t1"."enclosureTemperature", "t1"."offsetTemperature", "t1"."offsetBedTemperature", "t1"."offsetEnclosureTemperature" FROM "spo_spoolmodel" AS "t1" ORDER BY "t1"."displayName" DESC LIMIT ? OFFSET ?', [25, 0])
2021-05-29 08:28:58,303 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-05-29 08:28:58,304 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'password': 'illO', 'name': 'SpoolDatabase', 'host': 'localhost', 'fileLocation': u'/home/pi/.octoprint/data/SpoolManager/spoolmanager.db', 'user': 'Olli', 'useExternal': False, 'type': 'sqlite', 'port': 5432, 'baseFolder': u'/home/pi/.octoprint/data/SpoolManager'}
2021-05-29 08:28:58,309 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-05-29 08:28:58,313 - octoprint.plugins.SpoolManager.DatabaseManager.SQL - DEBUG - ('SELECT DISTINCT "t1"."vendor" FROM "spo_spoolmodel" AS "t1"', [])
2021-05-29 08:28:58,318 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
[... cut ...] cost", "t1"."costUnit", "t1"."labels", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml", "t1"."material", "t1"."materialCharacteristic", "t1"."density", "t1"."diameter", "t1"."diameterTolerance", "t1"."colorName", "t1"."color", "t1"."flowRateCompensation", "t1"."temperature", "t1"."bedTemperature", "t1"."enclosureTemperature", "t1"."offsetTemperature", "t1"."offsetBedTemperature", "t1"."offsetEnclosureTemperature" FROM "spo_spoolmodel" AS "t1" WHERE ("t1"."isTemplate" = ?)', [True])

```

OllisGit, 2021-05-29:
Hi @moorsey,

please attach the orig. file, because there is something wrong with the text-encoding. Via copy & paste it is not possible to reproduce the issue.

moorsey, 2021-05-29:
CSV file - https://www.dropbox.com/s/2eiytk8ohlevqoh/SpoolManager-Updated.csv?dl=0

moorsey, 2021-06-02:
This was an encoding issue, opened in NP++, changed to UTF8 and replaced "£" that had gone to some odd symbol

closed 2021-06-02 by moorsey