OllisGit/OctoPrint-SpoolManager #183: Changing spool Mid-print while paused leads to undefined behavior

codingcatgirl, 2021-06-06:
When changing a spool mid-print, SpoolManager asks whether you want the usage so far to be commited to the previously selected spool.

It works well – unless you do it while the print is paused. For some reason, then, it changes the spool just like that, without asking. This causes a pretty big problem when you notice that it didn't come up and klick the new spool *again*. When you confirm "commit to the previously selected spool" then, since the new spool was already selected, it will commit it to *that* spool.

So, two suggestions from me:

- When selecting a spool mid-print, the dialog should pop up too, currently it doesn't.
- When changing a spool mid-print without having selected the desired behavior (in the dialog), it should be rejected  – right now it doesn't seem to do that if the print is paused.
- Lastly: When selecting a "new" spool that is actually the one that is currently selected, the dialog probably shouldn't pop up.

dimo414, 2022-02-07:
I've observed this issue as well. Maybe it's since been fixed, but I don't think this issue should be auto-closed.

OllisGit, 2022-02-16:
There was a bugfix (#217) in version https://github.com/OllisGit/OctoPrint-SpoolManager/releases/tag/1.6.0 
No dialog, if the printer is in pause-mode the currently extruded amount is subtracted from the spool.

SO, from my option this issue is fixed. If not, please report with more details (e.g. which version do you use)

closed 2022-03-29 by github-actions[bot]