OllisGit/OctoPrint-SpoolManager #186: [Request] - Multiple templates

MadeforMaking, 2021-06-22:
As far as I'm aware and have tested. You can only have 1 template or profile as mentioned in #2. 

Is there development for multiple templates or even a template for each vendor?

timn, 2022-01-10:
I just came across this as well and I was confused when I tried to create templates for, e.g., PETG and PLA.

Before trying SPM I used Filament Manager, that has profiles which make it really easy to collect generic information. I wanted to try SpoolManager as it seems to be more actively maintained. If SPM would enable to collect certain types of filament in one place as a profile, e.g., vendor, material, density, cost etc., then when creating a specific a spool you could select a profile and add additional information such as color, weight etc. That could be done in the settings to decouple it from the specific spool management on the dashboard. Just my two cents.

OllisGit, 2022-01-10:
Hi,
thanks for the input. I also rethink about an solution. The easiest way to implement is as follows:
- You can mark as many spools as you like as a template spool (current: only one is allowed)
- When you create a new spool you can select from a list of template spools

[image]

[image]

timn, 2022-01-12:
That would already be an improvement! What would help is to distinguish creating a template from creating a spool (or gray out fields not needed for a template once the checkmark is made). In the overview list for spools a filter for templates would also be helpful.

OllisGit, 2022-03-18:
Hi, 
new Version 1.7.0 is released.

Please test and give feedback.

Thx, in advance
Olli

closed 2022-04-29 by github-actions[bot]