OllisGit/OctoPrint-SpoolManager #188: Conflict with TouchUI plug-in

gobbledygoober, 2021-07-05:
There appears to be a big related to Octoprint installations that have SpoolManager and TouchUI plugins installed. 

The issue was originally reported on the TouchUI GitHub, but the author marked the issue as Closed because they could not address it within their codebase:

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/500#issuecomment-874268401

If both plugins are installed, on the touch interface, links that are visible only after scrolling are unclickable. 

By process of elimination, disabling the SpoolManager plug-in allowed the touch interface to work as expected. The author suggested that:

> I have tracked down the code that is interfering. Inside SpoolManager there is a plugin named 'pick-a-color.js'. This piece of code called executeUnlessScrolled (its even in the name) is preventing the items to be clicked after scrolling on a touch device.

Wondering if this can be easily addressed on the SpoolManager side. Thanks for reading.

OllisGit, 2022-01-07:
I was able to reproduce the issue. In the next release your suggestion will be im implemented. Thx, for analysing!!!

OllisGit, 2022-01-12:
Hi,
in the latest version the issue should be fixed.
Please test and give a feedback.

Thx in advance
Olli

closed 2022-02-24 by github-actions[bot]