OllisGit/OctoPrint-SpoolManager #191: New feature: PTFE tube length(mm/gr) field to be added to check if the remain grams in the spool is enough

ldiegos, 2021-07-29:
Hi OllisGit,
     I was wondering if is it possible to add a new global parameter to fill the ptfe length or calculated weight, so the spool manager could warn if the amount of remaining filament is going to be enough to end the print.

In one of my last prints i forget to check the remain grams in the spool with the reality of the tubbing and the slicer. So the print fail because of the last grams(45cm of the tube).

The spool manager could take into account that the real grams of the print has to be added the lenght of the ptfe tube and for example some mm more to be able to the extruder to unload when the spool is empty. 

I don't know if this is going to be a pain to develop it, but please give a try.

Thank you so much.
Luis.

OllisGit, 2022-02-17:
Hi @ldiegos ,

in the latest  RC-Verison  1.7.0rc1 I changed a lot of stuff, maybe this issue is also solved.
Please test and give feedback

Thx, in advance
Olli

OllisGit, 2022-03-18:
Hi, 
new Version 1.7.0 is released.
[image]

Please test and give feedback.

Thx, in advance
Olli

closed 2022-04-29 by github-actions[bot]