OllisGit/OctoPrint-SpoolManager #192: Feature Request: Add ability to integrate with Filament Scale Enhanced Plugin

paintballkev812, 2021-07-31:
Would be nice to integrate with the [Filament Scale Enhanced Plugin](https://plugins.octoprint.org/plugins/filament_scale/). Ideally, I think you could get the actual scale reading pre and post print to get the actual used weight as an alternative to the calculated weight.

OllisGit, 2022-01-07:
fyi: i made some progress...the spoolholder is currently printing:
[image]

paintballkev812, 2022-04-20:
Awesome! Still keeping my eye out for this feature.

GaryGJR, 2023-03-29:
I'd like to upvote this feature.
Please add instantanious reading support for
     https://plugins.octoprint.org/plugins/filament_scale/ 

Note:  This reading is already on the Octoprint local webpage... from the plugin.

Here's my use case:
     I'm running Octoprint to a Prusa.  
My spool holder is sitting in a Sterilite Sealed Plastic Storage Bin.
A MakerHawk Digital Load Cell Weight Sensor HX711 AD Converter Breakout Module 5KG 
is attached to the bottom of the Storage Bin, and sits under the spool holder.

   
 See: https://www.amazon.com/MakerHawk-Converter-Breakout-Portable-Electronic/dp/B07GQPV6C4/ref=sr_1_2?crid=G2VS4NDJEBZM&keywords=HX711-based+load+cell&qid=1680065789&sprefix=hx711-based+load+cell%2Caps%2C92&sr=8-2 

So, for $12 with the existing Octoprint installed on a Raspberry Pi,  I have continuous readouts of filament on spool... as the print is happening, and as each new or used spool is loaded into the spool holder.

Now, For each part file uploaded to Octoprint I've added in the title My expected part weight in grams (from Prusaslicer).
    Note: Currently grams weight is sometimes hard to read on Octodash - my too long titles.

So, the current process is to 
   1. load the desired filament, 
   2. select the desired model on Octodash.  Note the expected part weight in grams.
   3. go to the local Octoprint website on my phone.  Note the measured wieght of filament remaining in grams.
   4. IF there is enough filament weight available, THEN print the part.

The desired change is to:
   1. Load the desired filament.
   2.  Read the instantaniously updated weight of remaining filament on the spool  (in grams) off the Octodash screen.
   3.  Select a model from Octodash that will run in the available filament, or change or mary spools.
   4.  Run the model.

Please add this feature.   
with   Filament Scale Enhanced , $12,  and some minor wiring,  the need for software generated guesses as to remaining filament on spools, and the need for a database to keep track of partial spools is mostly gone.

[image]

Thanks.

I'm Loving Octodash.  Using it daily.