OllisGit/OctoPrint-SpoolManager #193: Sorting by remaining doesn't seem to work

speedysk1, 2021-08-03:
I noticed that sorting by "Remaining" doesn't seem to work.  I don't know what it's sorting on specifically, but the Used % that I see on the screen certainly are not ordered correctly.  See attachment example
[image]

speedysk1, 2021-10-21:
@OllisGit Here you go.  Let me know if you need anything else.

[spoolmanager.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/7392188/spoolmanager.zip)

OllisGit, 2022-01-12:
Hi @speedysk1 ,
in the latest version the issue should be fixed.
Please test and give a feedback.

Thx in advance
Olli

closed 2022-02-23 by github-actions[bot]