OllisGit/OctoPrint-SpoolManager #196: Confirmation dialog shows wrong data

f-hnt, 2021-08-29:
The confirmation dialog, which pops up before starting a print shows wrong amount of filament.
This is valid for every spool. 
[image]

Octoprint restart does not affect the behavior. 

Plugin version: 1.4.3
OctoPrint 1.6.1
Python 3.8.10

sflebrun, 2021-09-03:
I am also seeing this problem.  The weight shown in the confirmation dialog is the weight of the spool when OctoPrint starts.  I am assuming that is when the Spool Manager plugin loads.  The database is correctly updated.  When a print is finished, the weight of the spool is the starting weight minus the amount used.  My guess is that the spool weight variable used by the confirmation is not being updated when the database is updated nor does it access the database to get the current value.  Once, I manually updated the weight of a spool and the new value did not show up in the confirmation dialog until I restarted OctoPrint.

Plugin: 1.4.3
OctoPrint:  1.6.1
Python: 3.7.3

Seppideluxe, 2021-09-15:
I can confirm the issue and the workaround. 

Version 1.7.0rc3
Python 3.7.3
Plugin: 1.4.3

At the moment I have not done any in-depth SQL debugging..

Seppideluxe, 2021-10-13:
as described once in the edit mode and then save.  then it updates the value.  but has to be done every time :-(

f-hnt, 2021-10-31:
> 
> 
> Hi, in latest release 1.5.0 I change the behaviour.
> 
> Please try and give me a feedback. Olli

Behavior does not changed after updating to 1.5.0. 
I also update to OctoPrint to 1.7.0 and it is the same. 

Could it be something in my installation?

dStankiewicz, 2021-12-07:
Same issue for me. Spool Manager 1.5.0 and OctoPrint 1.7.2. During start of a print, dialog box shows spool weight as entered initially, not updated with current data.
[image]

OllisGit, 2022-01-12:
Hi,
in the latest version the issue should be fixed.
Please test and give a feedback.

Thx in advance
Olli

f-hnt, 2022-01-17:
The issue seems to be fixed. :)

Thank you.

closed 2022-01-17 by f-hnt