OllisGit/OctoPrint-SpoolManager #202: [Feature request] allow entry of remaining spool weight and calculate corresponding remaining weight and length

mpnortn, 2021-09-27:
I love your plugin, thanks.  I have a scale to measure the weight of a spool+remaining filament. It would be very convenient to be able to enter this weight and have SpoolManager calculate the remaining filament weight (by subtracting the spool weight) and then calculate the corresponding remaining filament length using diameter and density. This would be a big time-saver if possible as I currently have to do these calculations manually in order to enter weight of filament used and used length.

Thanks!

OllisGit, 2022-01-07:
related to PR #206

OllisGit, 2022-01-12:
Hi @mpnortn! 
The mentioned PR is now in the latest release!

Because I have a lot of issues related the weight calculation I close this issue to get a more "up to date overview" in the issue-tracker.
If you found an issue in the latest version feel free to raise a new issue!

closed 2022-01-12 by OllisGit