OllisGit/OctoPrint-SpoolManager #212: Move from "Filament Manager"

WimVDK, 2021-11-02:
Just found this plugin which seems to be the successor to your "Filament Manager" that you took over from someone else a while ago. I swapped to that one as my android app i was using to manage spools didn't sync my spools back. I have exported my spools from there but i'm having problems with this plugin.

I found out where to add spools, but in order to add spools i need to add a vendor. The problem is i can't find out wher to add a vendor nor can i find out where/how to import my "Filament Manager" profiles & spools.

I'm getting a bit overwhelmed atm as i'm seeing "Printer Display" in the bottom left, a warning about "filament tracking is possible in the top right" and something about layer height etc not being detected. From what i get so far the "Printer display" is the same as what i'm seeing under "printer notifications". I don't need a M117 notification about the % or layer as i can see the % in the progress bar of my CR-10 Max.

Nor is the documentation any help atm,

i'd be very grateful of any and all assistance.

OllisGit, 2021-11-03:
Hi @WimVDK,
well documentation is always helpful ;-) ...Unfortunately it also took a lot of effort to create.
It is on my list (a starting point is a FAQ in the Wiki)..or feel free to provide a documentation which I could include here ;-)

But you are mixing multiple plugins . Each plugin has multiple responsibilities and some plugins interact with each other. But it is also possible to use each plugin individualy.

E.g. you are using DisplayLayerProgress which include theses features:
- "Printer-Display" in lower  corner. Every M117 which was sent by OP is displayed also in the Web-Browser. Can be deactivated via Plugin-Settings
[image]
- Extract Layer-Informations from the GCode during the upload of the file. If the gcode doesn't include special layer-comments the plugin could not extract this information and shows a warning (could also be disabled)

PrintJobHistory-Plugin
- This plugin captures each printjob-data and stores the result in a database. Beside the "out-of-the-box" information like start/end-time it captures also the layer-informations provided by the DLP-Plugin (if installed)
- And it could also capture the filament usage. But to do this, it needs a plugin that could capture the filament. The popup is a hint to select a tracking plugin 
[image]
[image]

 Switching from FilamentManager to SpoolManager
- The easiest way is to install both plugins and to export the FM-Data via the SpoolManager-Plugin. This way all data is exported in a format that could be easily manipulated via Excel (or Texteditor) and reimported into SpoolManager
 [image]

In the SpoolEdit-Dialog enter your spool vendor name into the field, hit enter and press save. After that, you can reuse the vendor name in the next spool.

Profiles like in FM is not supported in SpoolManager.

I hope I answered all of your questions...if so, please close the issue.

WimVDK, 2021-11-04:
One thing at a time, lets start by getting SpoolManager to work. I thought i might be mixing things but they 3 of those got installed cause SM recommended PJH and PJH recommended DLP in one way or another. I thought i had to export the csv from FM THROUGH FM, apparently not. So i managed export its db though SM, didn't see the sample csv so just straight up loaded it in SM but the "upload & import" is spinning. Could it be because there's a print going on?

OllisGit, 2022-01-07:
fyi: In the next release the import function is only available if the printer is not busy.

OllisGit, 2022-01-12:
Hi,
in the latest version the issue should be fixed. 
Please test and give a feedback.

Thx in advance
Olli

closed 2022-02-23 by github-actions[bot]