OllisGit/OctoPrint-SpoolManager #214: Can't figure out why

Filmman586, 2021-11-09:
I'm getting this message, but I dont know why...Any suggestions?

[image]

Lou

MrRexRuthless, 2021-11-12:
I just started receiving this as well.  Started a print this morning from Octodash, i believe and it failed.  Went to restart again from the OctoPrint Web Interface and received this error.  Have restarted PI.

OllisGit, 2021-11-14:
Hi @Filmman586,

to calculate the needed weight the plugins needs some data from OP. The data is stored in a file called ```.metadata.json``` and is located in the upload folder (look for analysis/filament). 
```
        "analysis": {
            "printingArea": {
                "maxX": 127.881,
                "maxY": 95.506,
                "maxZ": 10.9,
                "minX": 88.081,
                "minY": 42.554,
                "minZ": 0.3
            },
            "dimensions": {
                "depth": 52.952,
                "height": 10.6,
                "width": 39.8
            },
            "estimatedPrintTime": 2754.571579330689,
            "filament": {
                "tool0": {
                    "length": 2016.75497,
                    "volume": 0.0
                }
            }
        }
``` 
If this meta data is not available the popup appears (or you didn't select a spool for tool 0).
Please check which option do you select in the features-section of the settings.
[image]

MrRexRuthless, 2021-11-15:
> I just started receiving this as well.  Started a print this morning from Octodash, i believe and it failed.  Went to restart again from the OctoPrint Web Interface and received this error.  Have restarted PI.

Mine seemed to go away by itself.  I will update if I see it repeat.

FoxRomeo, 2022-02-08:
I had the same issue.
With the data here I could find out more data and the cause (in my case).
Hardware: RPi3

It happened in my case only with large prints that have no estimatedPrintTime when the print starts. Then I receive the "no spool selected" dialog.
As soon as I installed other plugins that make faster estimate calculations, then this dialog disapears.

Therefor it seems to be no error in this plugin, more like missing required data that leads to a (maybe missleading) dialog.

If you encounter such beavior check for the estimate time, and try the plugins OctoPrint-MetadataPreprocessor and OctoPrint-PrintTimeGenius could help you in case (it did in mine).

closed 2022-03-21 by github-actions[bot]