OllisGit/OctoPrint-SpoolManager #215: Database Manager "Spam" in Log File

EvilBlood, 2021-11-14:
Hi, 
the DatabaseManager creates a lot of log entries.

`2021-11-14 17:05:41,422 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,425 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,457 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,458 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,459 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,469 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,470 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,473 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,499 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,500 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,501 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,509 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,509 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,510 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,516 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,516 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,517 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,522 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,522 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,523 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,529 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,529 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,531 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,544 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,544 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,545 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:41,547 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:41,548 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:41,551 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:42,786 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:42,787 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:42,791 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:42,801 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:42,802 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:42,805 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:42,833 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:42,834 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:42,836 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:42,843 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:05:42,846 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:05:42,847 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-11-14 17:05:42,853 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
[... cut ...]
2021-11-14 17:07:25,818 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-11-14 17:07:25,818 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-11-14 17:07:25,822 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions`

OllisGit, 2021-11-14:
Hi @EvilBlood,
yes it is annoying and thanks for giving a short note about that ;-)

BR
Olli

Gurkengewuerz, 2022-01-04:
Any updates on this?

OllisGit, 2022-01-12:
Hi,
in the latest version the issue should be fixed. Only visible if SQL logging is enabled.
Please test and give a feedback.

Thx in advance
Olli

EvilBlood, 2022-01-18:
Fehler ist behoben danke :)

closed 2022-01-18 by EvilBlood