OllisGit/OctoPrint-SpoolManager #216: SpoolManager together with FilamentManager  produces error in filament sidebar usage

Mascherino, 2021-11-15:
Hey I don't know long this error has been present, but I only recently noticed it.

Using only SpoolManager displays the filament usage in the sidebar in meters as expected.
Using only FilamentManager  also works as expected, showing filament usage in grams and meters in the sidebar.
But when using them both together the filament usage displays only display "NaNm" and I get the following error in the developer console.

[image]

[image]

I'm using Octoprint 1.7.0, SpoolManager 1.5.0 and FilamentManager 1.9.0.
I have disabled all other plugins to avoid any conflicts with other 3rd party plugins.

I'll just use FilamantManager only from now on as it seems to be working just fine.

OllisGit, 2022-01-12:
Hi @Mascherino ,
in the latest version the issue should be fixed.
Please test and give a feedback.

Thx in advance
Olli

Mascherino, 2022-01-13:
Combination of both plugins works fine again. Thank you.

closed 2022-01-13 by Mascherino