OllisGit/OctoPrint-SpoolManager #218: ERROR MESSAGE when printing

chemixt, 2021-11-22:
I recently updated to pytho 3 on my pi, and started having errors when printing coming from Spool manager.  Here is the message:

[image]

It does not seem to be tracking usage any more, and when I try to edit the data on my spools page, it shows no spools even though I have "all" checked, but if I click on "all" then my spools show up. 

I have not tried to uninstall it because I do not want to lose my data, but if i need to I can do so.

jonsag, 2021-11-22:
Same message for me.
Have been runnning python3 for a while, so I don't think it's related to that.

MrRexRuthless, 2021-11-23:
I had the same issue, open a couple of the last used spools and rebooted the PI and everything resolved itself, also, I had a spool that I had used "more" than the spool total.  Zeroed that out.

Not a fix of course but I was able to resolve the issue

chemixt, 2021-11-24:
Thanks for the help.  I tried editing and rebooting.  I found a spool in the data that had gone negative (I am not sure if i entered the used amount correct or actually used that spool without changing it to the real one being used.  But after a reboot of the Pi. I still have the same problem.

chemixt, 2021-11-25:
I pushed back to the previous version [1.4.3) today, and problem went away and it seems to work fine.   My problem seems to be related to an error message that indicated that no spools have been selected and there is no meta data.

OllisGit, 2022-01-07:
Hi @chemixt,
the hint regarding "metadata" means that OctoPrint was not able or not already provide the needed metatdata for the selected model to calculate the needed amount of filament.

Each gcode file has a corresponding metadata entry in the file ```<op>/uploads/.metadata.json```.
Looks like this:
```json
  "3DBenchy.gcode": {
    "DisplayLayerProgress": {
      "totalLayerCountWithoutOffset": "159"
    },
    "analysis": {
      "estimatedPrintTime": 3529.315478428196,
      "printingArea": {
        "maxZ": 48.05,
        "maxX": 119.791,
        "maxY": 122.755,
        "minX": 20.0,
        "minY": 50.0,
        "minZ": 0.3
      },
      "dimensions": {
        "width": 99.791,
        "depth": 72.755,
        "height": 47.75
      },
      "filament": {
        "tool0": {
          "volume": 10.928834037976793,
          "length": 4543.68119999991
        }
      }
    }
``` 
There is a strategy when to create such data, please verify the plugin-settings ("Feature"):
[image]

If the issue is still present, please attach the octoprint.log, maybe there is a different issue.

chemixt, 2022-01-08:
Thanks for the tip.  Sorry I do not understand. Where should I find this
feature setting?

OllisGit, 2022-01-12:
[image]

chemixt, 2022-01-15:
Thanks!  Not sure how I missed that

closed 2022-01-16 by OllisGit

chemixt, 2022-04-10:
Sorry, I lost the link to this thread.  I am still having the same issue with my Makerbot Replicator 2x.  I can no longer track any filament usage.  everything seems to be set correctly but even when I select a valid spool, I get this message.  I thought I figure it out because all my spools were checked as invalid on the spool editor. so I unchecked them all and I still get the same message. 

octopi.local says:

ATTENTION: Needed filament could not calculated (missing metadata or spool-fields)

There are no spools selected for the following tools dispite them being used (maybe) by this print:
- Tool 0
- Tool 1
Do you want to start the print without selected spools?**_

I checked my feature settings and they are as you had indicated

[image]

Here is my octoprint log if it helps. 

[octoprint.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/8460171/octoprint.zip)

chemixt, 2022-04-11:
I tried to roll back to 1.4.3 again, but the download link does not work.

chemixt, 2022-04-22:
Ok, sorry, I figured out how to roll back to 1.4.3, and it works with my makerbot Replicator 2x again.  I Hope you are able to get that latest version to work with it.,  I like the updates that you made, but with my system, it no longer tracks usage and gives me an error whenever I print.  
ATTENTION: Needed filament could not calculated (missing metadata or spool-fields)

Error message:
There are no spools selected for the following tools dispite them being used (maybe) by this print:

Tool 0
Tool 1
Do you want to start the print without selected spools?**_