OllisGit/OctoPrint-SpoolManager #219: Filament tracking not working

bambuleee, 2021-11-30:
Spool is selected for printing.
[image]

But in the finished print dialog it's not recognized. Stop working from one day to another.
Reinstalled PrintJobHistory plugin and Sppol Manager.

[image]

Tried with different spools.

OllisGit, 2022-01-07:
Hi @bambuleee,
please attach the octoprint.log. Maybe there are some  error entries.

bambuleee, 2022-01-08:
Hi,this is my latest log.
I think the spool manager is tracking, but the information is not passed through to the print hiostory.

`2022-01-07 09:26:31,456 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2022-01-07 09:26:31,459 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2022-01-07 09:26:31,459 - octoprint.server - INFO - OctoPrint 1.7.2
2022-01-07 09:26:31,461 - octoprint.plugin.core - INFO - 40 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Autologin Configuration (0.2.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_autologin_config
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Backup Scheduler (0.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_backupscheduler
|  BLTouch Plugin (0.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_BLTouch
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Creality-2x-temperature-reporting-fix (0.0.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ender3v2tempfix
|  Cura Thumbnails (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ultimakerformatpackage
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/eventmanager
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.10.11) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  InlineConfirm (1.1.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_inlineconfirm
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Marlin GCode Documentation (0.14.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_marlingcodedocumentation
|  OctoDash Companion (0.0.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octodashcompanion
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.10.28) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintJobHistory (1.15.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory
|  PSU Control (1.0.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_psucontrol
|  PSU Control - Tasmota (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_psucontrol_tasmota
|  Pushover (0.3.8) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pushover
|  Resource Monitor (0.3.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_resource_monitor
|  Slicer Print Time Estimator (1.1.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SlicerEstimator
|  Slicer Thumbnails (1.0.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  SpoolManager Plugin (1.5.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager
|  UI Customizer (0.1.8.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_uicustomizer
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WebDAV Backup (0.4.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_webdavbackup
|  Webhooks (3.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_webhooks
|  WiFi Status (1.7.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_wifistatus
|  Z Probe Offset (0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_z_probe_offset
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2022-01-07 09:26:31,469 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|    cores: 4
|    freq: 1500.0
|    ram: 1909063680
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  plugins:
|    pi_support:
|      model: Raspberry Pi 4 Model B Rev 1.1
|      octopi_version: 0.18.0
|      throttle_state: '0x0'
|  python:
|    pip: 20.3.3
|    version: 3.7.3
|    virtualenv: /home/pi/oprint
2022-01-07 09:26:31,469 - octoprint.server - INFO - ------------------------------------------------------------------------------
[... cut ...]
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 372, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octodashcompanion/__init__.py", line 39, in on_settings_load
    with open(self.config_file) as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.config/octodash/config.json'
`

bambuleee, 2022-01-08:
Hi,this is my latest log.
I think the spool manager is tracking, but the information is not passed through to the print hiostory.

[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/7832051/octoprint.log)

OllisGit, 2022-01-08:
Hi @bambuleee,

the log file does not cover a print job. Do you have a log file where a print job run is logged?

Btw. there is something wrong with the OctoDash-Config
```
2022-01-07 13:18:16,453 - octoprint.plugins.octodashcompanion - INFO - attempting open of /home/pi/.config/octodash/config.json.
2022-01-07 13:18:16,454 - octoprint.server.api.settings - ERROR - Could not load settings for plugin OctoDash Companion (0.0.6)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 372, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octodashcompanion/__init__.py", line 39, in on_settings_load
    with open(self.config_file) as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.config/octodash/config.json'
2022-01-07 13:18:17,028 - octoprint.plugins.octodashcompanion - INFO - attempting open of /home/pi/.config/octodash/config.json.
2022-01-07 13:18:17,028 - octoprint.server.api.settings - ERROR - Could not load settings for plugin OctoDash Companion (0.0.6)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 372, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octodashcompanion/__init__.py", line 39, in on_settings_load
    with open(self.config_file) as file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.config/octodash/config.json'
```

bambuleee, 2022-01-08:
Hi,
sorry. Here is a logfile including a printjob.
OctoDash is fixed. It was not installed anymore. Thank you. ;
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/7834163/octoprint.log)
)

bambuleee, 2022-01-10:
> Hi @bambuleee,
>
> the log file does not cover a print job. Do you have a log file where a print job run is logged?
>
> Btw. there is something wrong with the OctoDash-Config
>
> ```
> 2022-01-07 13:18:16,453 [...]

Hi,
sorry. Here is a logfile including a printjob.
OctoDash is fixed. It was not installed anymore. Thank you. ;)
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/7839763/octoprint.log)

OllisGit, 2022-01-12:
In the log I can see that the tracking of the filament works:
[image]
But it looks like the PJH could not read the data. I will move this issue the PJH repository.