OllisGit/OctoPrint-SpoolManager #222: Integrate metadata generated by Slicer Estimator

NilsRo, 2022-01-10:
Hi.

I am the developer of Slicer Estimator Plugin (https://github.com/NilsRo/OctoPrint-SlicerEstimator) and like to discuss if it would be helpful to read out the material of a file uploaded to OctoPrint. This would be a simple solution between manual selection and the barcode reader.

The plugin made it possible to add read custom metadata from GCODE and add it to the file metadata for example the material name or Curas GUID. You can read it via and get a dictionary back. self._file_manager._storage_managers["local"].get_additional_metadata(<path of the file>,"slicer");

So it should be easy to add this optionally for automatic selection of the material. 

Cheers,
Nils

OllisGit, 2022-01-12:
Hi @NilsRo,

I already tested your plugin and it is really nice.
[image]

Your plugin can/should be included into SpoolManager and also into PrintJobHistory-Plugin. 
In PJH currently all key-value comments were captured (see https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/Slicer-Settings), but it is maybe a better approach to capture only dedicated keys like ```;Slicer info:material_guid;Material GUID;{material_guid}```

Regarding SpoolManager-Plugin I received  a couple of request for automatic spool selection (see https://github.com/OllisGit/OctoPrint-SpoolManager/labels/FEATURE%3A%20FILAMENT%20SELECTION)

I am quit unsure about the priority of such feature...currently it is in the "long-term" column of my project plan, see https://github.com/OllisGit/OctoPrint-SpoolManager/projects/1.

There are a couple of smaller things in the short/mid-term plan, but my biggest "fear" is finalizing the external database-stuff..and this will take some time.

...but is "tracked", now ;-)

NilsRo, 2022-01-12:
Hi @OllisGit 

Nice to here. Many thanks. 😄 

With Curas OctoPrint Plugin it would be nice to have an automatic filament selection as print can be started directly from the slicer. So personally I does use OctoPrints GUI mostly to monitor the print longterm as Cura does support Webcam monitoring also or if I like to print something a second time. 

Both topics was the reason I spend some thoughts how to integrate (material) metadata. Curas direct print workflow is not supported in SpoolManager yet if I am right? Perhaps I have some time to contribute when travel by train next time. 🙂 

Cheers
Nils

OllisGit, 2022-01-12:
...train traveling time....5h to munich every 3-4 weeks....so much free time for programming, no phone calls...i missed that ;-(

"Curas direct print workflow", what do you mean with that? Of course you can send and auto-start the job directly from cura and then we need to talk about "selecting" an existing spool thru the SPM or should the "spool-data" (material, diameter,...) taken from gcode model.

- "selecting":  what should be the identifier, the "Display Name" from Cura? Should this value 1:1 with SPM Name or an other field in SPM "CURA Material Identifier"....identifier = Display Name + Brand + Material Type + Colorname
- "spool-data":  just reading some spool-data and add this the job-history entry
 
[image]

NilsRo, 2022-01-12:
Yes, Munich - Hamburg. Nice trip. ;-)

Extracting all data available and creating new spools would be nice but in my case simply selecting an existing spool would be perfect. 

Directly printing PETG...changing to PLA..printing next model without logging into OctoPrint GUI. That is my scenario...

identifier = Display Name + Brand + Material Type + Colorname

Everything necessary was added into Curas Placeholders after a discussion with the developer fieldofview in the OctoPrint forum. So if everything is named the same in SPM we are fine. Also useable as a generic solution with other slicers. 

Using Curas Spool GUID would be technically better but not generic in my opinion.

closed 2022-02-23 by github-actions[bot]