OllisGit/OctoPrint-SpoolManager #223: Plugin not working, because of Database issues. Check Plugin-Settings!

d4rken, 2022-01-13:
[image]

Octoprint Version 1.7.2
SpoolManager Version: 1.6.0

[octoprint-spoolmanager.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/7862273/octoprint-spoolmanager.log)
[spoolmanager.zip](https://github.com/OllisGit/OctoPrint-SpoolManager/files/7862279/spoolmanager.zip)

As far as I can tell, the plugin does work though and tracks consumed filament.

OllisGit, 2022-01-13:
Really strange...from implementation perspective it is not possible:
[image]
There is only one boolean flag that decide to show the error-sentence OR the table-data and NOT both.

Do you have other plugins installed that "improves" the UI? 
In your log I missed the starting phase, so I can't see what plugins are installed.

d4rken, 2022-01-13:
* Cura Thumbnails (1.0.1) 
* Dashboard (1.19.6) 
* DisplayLayerProgress Plugin (1.27.2)
* FileManager (0.1.6)
* Firmware Updater (1.13.0)
* Navbar Temperature Plugin (0.14) 
* Octolapse (0.4.1) 
* Preheat Button (0.8.0) 
* PrintJobHistory (1.15.2) 
* PrintTimeGenius Plugin (2.2.8) 
* Resource Monitor (0.3.6) 
* Slicer Thumbnails (1.0.0)
* SpoolManager Plugin (1.6.0) 
* Tab Order (0.5.12) 
* Themeify (1.2.2)

d4rken, 2022-01-13:
Seems to be caused by Themeify:
https://github.com/birkbjo/OctoPrint-Themeify

OllisGit, 2022-01-15:
Hi @d4rken,

I installed the plugin, but I am not able to reproduce the issue. 
So, when you disable "Themeify" then the DB message is gone, right?

What settings do you use?
[image]

[image]

d4rken, 2022-01-15:
I played a bit more and when I disable "customizations" and reload the page, it's also fixed.

[image]
[image]
[image]

OllisGit, 2022-01-16:
Thx, for the screenshot! Now I was able to identify the issue, it is related to ```.row-fluid```.

I will fix this in the next release. In the meantime you need to remove this entry...or ignore the message (like you do in the past).

BR
Olli

d4rken, 2022-01-16:
Thank you so much for all your contributions to the 3D printing community :heart:.

OllisGit, 2022-01-31:
Hi @d4rken,

the new bugfix release 1.6.1 is out.

Please test and give feedback.
Thx, in advance
Olli

d4rken, 2022-02-01:
Fixed :+1:

closed 2022-02-01 by d4rken