OllisGit/OctoPrint-SpoolManager #225: Unable to copy spool not currently in use.

daftada, 2022-01-20:
Unable to copy any spools when Octoprint is printing. I can understand locking the spool in use, but locking them all is prohibitive. I need to add inventory while waiting for prints to finish.  Copying is a real time-saver as I purchase multiple spools with only the colour differing. I am sure that I am not alone in this. Thanks.

OllisGit, 2022-01-24:
Yes, agree should be optimized ;-)

Note to myself: Don't do it like that, more details must be included.
```
self.printerStateViewModel.isPrinting()
```

OllisGit, 2022-02-17:
Hi @daftada,

in the latest  RC-Verison  1.7.0rc1 I changed a lot of stuff, maybe this issue is also solved.
Please test and give feedback

Thx, in advance
Olli

daftada, 2022-02-22:
Sorry for the tardy reply.   Yes, this appears to be working as expected
now.
Many thanks for correcting this.