OllisGit/OctoPrint-SpoolManager #226: Flow rate - additional granularity

bkhofmann, 2022-01-21:
The flow rate field appears to be limited to fixed 1% increments.  It would be useful to be able to have more granularity such as 0.1% increments to avoid the need for rounding.  I do realize that Marlin G-code M221 command also seems to be limited to 1% increments, but current case does not interface this field with the printer.

OllisGit, 2022-01-24:
Hi @bkhofmann,

changing the UI is not a big deal, but also on the control-tab (see below) you can only increment by +1 and not 0.1.

Currently the spool filed value is not send to the printer (as mentioned), but in future it should be possible.

[image]

So, it is not clear to me why should the user enter float-values when there is no way to send/interpret by the printer?

bkhofmann, 2022-01-24:
Initially, I was thinking since it was not being sent to the printer, its only usefulness would be to store data for the end user, some users might have "calibrated" flow with resolutions less than 1% and desired to store this data here.  I have since come to the understanding that for typical 1.75mm filament diameter tolerance ranges (+-0.03mm) the cross sectional area of the filament can vary by roughly +- 3.5% and for tight tolerance ranges (+-0.02mm) varies roughly +-2.3%, so "calibrating" flow with resolution much less than about 2% steps makes little sense, making this desire for capturing sub 1% granularity moot.  Thanks for humoring me, I'd like to rescind this issue/feature request based on this recently acquired understanding.

closed 2022-03-06 by github-actions[bot]