OllisGit/OctoPrint-SpoolManager #228: Hide inactive tools toggle doesn't work

MarloSM, 2022-01-24:
Hi, great work on the plugin!

Just noticed recently that the Hide inactive toggle acts as inactive if the page is refreshed even though it's ticked. Seems like the page defaults this setting to inactive, rather than taking the value of the variable? Haven't dived deep, just guessing.

**After refresh, before re toggle**
[image]

**After refresh, after re toggle**
[image]

OllisGit, 2022-01-24:
Side note to myself: 
- total spool count in tab not refreshed
- output spool-count to selection-dialog

MarloSM, 2022-02-19:
I had a look today on Google Chrome and Firefox and this issue is still around, sorry.

There's an added UI issue where if you click on the words "Hide inactive spools", it toggles "Hid empty spools". As if the active area for the first option is bigger. Only clicking on the tickbox for "Hide inactive spools" works.

OllisGit, 2022-03-18:
Hi @MarloSM, 
new Version 1.7.0 is released and fix the issue

Please test and give feedback.

Thx, in advance
Olli

MarloSM, 2022-03-29:
Hi @OllisGit. Working great now, thanks.

closed 2022-03-29 by MarloSM