OllisGit/OctoPrint-SpoolManager #231: SpoolManager not tracking usage

jet8300, 2022-01-30:
SpoolManager is not working correctly for me. It is not tracking filament usage even when a spool is selected. If I attempt to start a print with no spool selected, I also do not get a prompt telling me to select one, even though I have the options selected to verify a spool is selected.

I am running the following environment;

- OctoPrint 1.7.3
- Python 3.7.3
- OctoPi 0.18.0

Enabled 3rd party plugins:

- Bed Visualizer 1.1.1
- OctoKlipper 0.3.9.4
- PrintJobHistory 1.16.0
- Slicer Thumbnails 1.0.0
- SpoolManager Plugin 1.6.0

I have attempted uninstalling SpoolManager and fully reinstalling it. I even created a new spool rather than importing my old database.

Any help would be greatly appreciated. I love this plugin!

OllisGit, 2022-01-31:
Hi @jet8300,

interesting you love this plugin, but it is not working for you...my feelings would be hate and not love ;-)  

Please attach the octoprint.log maybe there is an error documented.

jet8300, 2022-02-05:
Well, it was working perfectly until recently. I think that the issue is actually more something to do with superslicer and octoprint, and not your plugin, the more I look into this. 
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/8007107/octoprint.log)

Octoprint doesn't appear to be getting the filament usage data from the gcode generated by superslicer.

jet8300, 2022-02-09:
So after more digging, it looks like Octoprint is getting the data from the gcode; it's listing it in the metadata. Not sure if that helps.

OllisGit, 2022-02-15:
Hi @jet8300,

the real extruded data (length) is captured during the print, by reading each  gcode line (e.g. G1 ... E1000).

Activate the "Extrusion debugging" in the plugin settings, then you can see during the print if the extrusion is captured.
[image]
[image]

Maybe it is related to your used slicer, so please attach the gcode as well, so I can try to reproduce the issue in my environment.

jet8300, 2022-02-18:
Here is an example gcode. I am using the latest version of superslicer.

[Bridge calibration_gcode.txt](https://github.com/OllisGit/OctoPrint-SpoolManager/files/8097915/Bridge.calibration_gcode.txt)

I ran this print and here is the octoprint log:
[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/8097925/octoprint.log)

I don't see any extrusion data in there, but I did enable debugging as you asked.

OllisGit, 2022-03-18:
Hi @jet8300,
I analysed your provided GCode and it looks like that the movement of extruder motor is interpreted as absolut positioning. But in the gcode-file all E-Values looks like relative values.

[image]

The default behaviour from OctoPrint and my plugin is, that if no special gcode for movement-mode is provided, each value is interpreted as absolut.
OP is not able to calculate the total length. 

From your gcode:
[image]

From OP (```.metadata.json```)
[image]

My Assumption is, that your printer acts the other way around. No special command, then it is relative.

G-Code for movement-mode are:
- G90 absolute
- G91 relative
- M82 absoute extruder
- M83 relative extruder

Try to add a ```M83``` command in your gode.

closed 2022-04-28 by github-actions[bot]