OllisGit/OctoPrint-SpoolManager #235: Update "color" only when empty

Jeanlucpica, 2022-02-07:
Observed behaviour:
I open an already created spool from the list to view it or update information. The spool has a custom color text value. Meaning the written color name does not match the hex code. This custom color name was saved when creating the spool. As soon as I open the spool with the custom color name, its value is overwritten if the hex code has a valid name. My custom data is lost when I save.
[image]

Expected behaviour:
Once entered, a field should keep its initial value. In case of the custom color name, it should remain as is if it is filled. Only when creating a new spool, the color name matching the color hex value should be filled. If a value is already stored it should remain untouched.
[image]

OllisGit, 2022-02-17:
Hi @Jeanlucpica,

in the latest  RC-Verison  1.7.0rc1 I changed a lot of stuff, maybe this issue is also solved.
Please test and give feedback

Thx, in advance
Olli

Jeanlucpica, 2022-02-18:
I just installed the 1.7.0rc1 and can confirm that this issue is fixed. All customer colors now remain as is. Thank you very much for fixing this so quickly 👍

closed 2022-02-18 by Jeanlucpica