OllisGit/OctoPrint-SpoolManager #236: Sporadically list of spools empty or missing spools

Jeanlucpica, 2022-02-07:
I have some strange behaviours with this plugin. Sporadically my list of spools will stay empty and show nothing, even though there should be several spools be displayed. Often it looks like this when I open the plugin page:
[image]
A forced refresh usually then shows the complete list of spools.

But a more bothering issue happens especially after adding new spools or editing existing ones. As soon it goes back to the list suddenly some spools disappeared from the list. I can't really tell why some are shown afterwards and why some are not. Here is an example where 10 spools should actually be displayed, but I only see 7:
[image]

I already tried disabling some other plugins I have in use, but could not link this problem to a compatibility issue with another plugin. It seems to happen independently. Maybe somebody else has the same problem? Or can you give some advice on why this is happening?

Some technical details:
- Windows 10
- Microsoft Edge 98.0.1108.43
- OctoPrint 1.7.3
- OctoPi 0.18.0

OllisGit, 2022-02-16:
Hi @Jeanlucpica,

regarding the first issue "empty list", pease take a look into the octoprint.log. Maybe the server could not load the spool infos from the database. Also, please open the "Browser-Developer-Tools (press F12)". In the console and on the Network-Tab there should be no red-lines.

In your screenshot is a color-filter active (show only 9 colors), maybe that is reason why you can't see all spools.
But I optimized the total-output in the next release.
[image]

OllisGit, 2022-02-17:
Hi @Jeanlucpica,

in the latest  RC-Verison  1.7.0rc1 I changed a lot of stuff, maybe this issue is also solved.
Please test and give feedback

Thx, in advance
Olli

OllisGit, 2022-03-18:
Hi,
new Version 1.7.0 is released.

Please test and give feedback.

Thx, in advance
Olli

Jeanlucpica, 2022-03-19:
Hi,
I installed version 1.7.0 and added a new spool. I can't really say it fixed things, but I believe I can give some more details what I am observing.

1. The list is always empty when switching to the "Spools" tab after a reboot of my OctoPi. If I hit the reload button everything comes up as expected. Not a problem, just inconvenient.
[image]

2. I do still have some unexpected behaviour when adding/modifying a spool which led to the initial report here. I add a new spool like that:
[image]
First and last use are empty. I save the spool and I am back in the list. If I will now select a different spool which has been used before, it will show me all the details including the date for last and first use. Also fine. **But If I will now close that spool and then open the new spool, which was never used, it now will display the last and first use date of the spool I just opened before.** So it mixes up data. And if you do not pay attention and save it like this, you will save wrong data to the database.

So number 2 is an actual bug and I can reproduce it. It also happened to me that suddenly the color filter was set and excluded the color of the spool I just had added. However I am unable to reproduce it but it happened also with 1.7.0. Maybe you want to check if the problem of 2 maybe also applies to other fields and may cause this issue.

I hope this info helps to resolve this.

closed 2022-04-29 by github-actions[bot]