OllisGit/OctoPrint-SpoolManager #237: Installation Issue Spool Manager 1.6.1

chiral-lab, 2022-02-08:
Hi,
when I try to install the current spool manager on Octoprint Version 1.7.3 the installation log runs fine, installs all dependecies and finishes with an ok message. But in the upper right corner of Octoprint shows "unknown" installed and a restart is necessary. After the restart the spool manager is not present, neither in its function nore in the plugin list. This happens when I install with the plugin repository or the git link.

Please let me know if you need other information.

best regards
Chiral

OllisGit, 2022-02-09:
Hi @chiral-lab,

please attach the log files from OP or the "Systeminfo Bundle" (Settings-Menu in the bottom)
[image]

chiral-lab, 2022-02-10:
Hi,

sorry that I forgot the logs. They are attached to my comment now. I tried to install Spool Manager again just before saving the logs so the interesting parts should be near the end.

best regards
Chiral 

[octoprint.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/8038634/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/OllisGit/OctoPrint-SpoolManager/files/8038635/plugin_pluginmanager_console.log)

OllisGit, 2022-02-16:
hmm..I use the same python and database lib (peewee) version...at the moment I have no idea why the this happened.

Please take a look into this file: ```/home/pi/oprint/lib/python3.7/site-packages/peewee.py```
This file should include this class ```class CharField(_StringField):```

chiral-lab, 2022-02-17:
When I look with nano into the peewee.py under the described path I am either doing something wrong or the file is empty.

[image]

OllisGit, 2022-02-17:
Okay, thats the issue. There was something wrong during installation of the 3rd party lib peewee.
You can also check the size of the file. 

You can try to install the lib manually
```
cd /home/pi/oprint/bin
./pip uninstall peewee
./pip install peewee
```

Then restart your server and hopefully the error is gone!

chiral-lab, 2022-02-18:
Thanks.
Well the problem seems to be my system. Even pip is not installed. installing pip manually does not work and installing peewee with pip3 does not solve the issue. Maybe I broke something when I switched from Python 2 to Python 3.

I'd say I close the issue since it seems to me the plugin is not the root cause.

I'll set up a new Octoprint when I find the time and will try again to install the Plugin.

closed 2022-02-18 by chiral-lab