OllisGit/OctoPrint-SpoolManager #239: Design the Spool Manager to accommodate multiple spools of the same type ( Manufacturer / Vendor / Material )

jDavidnet, 2022-02-08:
I find my self buying again spools I liked, and it would be nice to be able to just add an additional spool of that type, and to have them grouped together.

It would be a bonus to be able to track the end of a spool, and the beginning of a 2nd one in that set.

OllisGit, 2022-02-16:
Hi @jDavidnet,

sounds like an interesting idea. 
It would be really helpful if you can make a suggestion how the UI should handle this feature. Hand Drawing, HTML-Mockup...are welcome ;-)

jDavidnet, 2022-02-16:
I say start simple.  Allow someone to duplicate and link certain values.
V0 would just be a variation of the template button, saying "add another roll," or something, "maybe duplicate roll"

Later on maybe it would involve more linking features, like updating various fields on one would update on both, or you might have a history list connecting all of the rolls of a type together, maybe all Amazon Basics White could be listed below, with each of their filament roll stats like meters / grams remaining, last used, and first opened date.

maybe you might also try to remind someone if you have a roll of the same time with less on it, to us that first?  maybe that's a setting?

OllisGit, 2022-03-18:
Hi @jDavidnet, 
new Version 1.7.0 is released and include the feature "Multiple Templates" maybe this could help.
(was already included in RC1: https://github.com/OllisGit/OctoPrint-SpoolManager/releases/tag/1.7.0rc1)

Please test and give feedback.

Thx, in advance
Olli

closed 2022-04-28 by github-actions[bot]