OllisGit/OctoPrint-SpoolManager #241: Feature Request: Support Multicolor Filaments (ex. Amolen Color Changing, MatterHackers Quantum)

jDavidnet, 2022-02-10:
As Filaments get more creative, they no longer are just one color.  Some filaments slowly change color via a sequence of colors, like Amolen's color changing filaments.  Some are split with one color on the left of the filament and another on the right like MatterHackers Quantum.  Others are glow in the dark, so they are 1 color in daylight and a different color in the dark.  Yet others change color depending on the temperature.  The spool manager should allow you to enter those colors, either as "other" or "multicolor" and allow a description space, or allow each filament to have a color list.  Maybe spools should allow for an image of the spool from the vendor?

OllisGit, 2022-02-16:
I understand your demand, but I am not a big UI-Expert..so I was more then happy to find an "out-of-the-box" color picker that works with the old css bootstrap version from octoprint.
I think to find a "gradient-color-picker" is also not very easy.

Adding an Image to the Spool was on my list, but first I want to finish the "external database" feature....and adding an image to the database is also a challenge, see https://community.octoprint.org/t/best-approach-for-storing-images/20935/2

jDavidnet, 2022-03-19:
can you add this to the backlog?

closed 2022-06-02 by github-actions[bot]