OllisGit/OctoPrint-SpoolManager #246: Support for MQTT

notenoughtech, 2022-02-16:
Hi, 

I wonder if it would be possible to add the MQTT option to share the writes to the database? It would enable logging externally without reading the DB or hosting it on another server. 

Thanks

OllisGit, 2022-02-17:
Hi @notenoughtech,

in combination with this plugin [MQTT](https://github.com/OctoPrint/OctoPrint-MQTT) and the checkbox "Unclassified events" , it should be easy to implement. I just need to fire an event.

These are the events that could be helpful:

- updateSelectedSpoolWeight
- selectedSpool
- removeSelection
- updatedSpoolData
- addedSpool
- deletedSpool

Did I miss something?

notenoughtech, 2022-02-17:
I couldn't see it on the list of events so I skipped it.
I wrote a small python scraper last night that hooks up directly to the SQL
file and sends the updates.

I'll think about the event approach now as well. I want to use some NFC to
inject the spools to the database. Are you anticipating any issues caused
by duplicating DB entries?

OllisGit, 2022-02-17:
Hi @notenoughtech,

in the latest RC-Version 1.7.0rc1 I implemented the event-mechanism.
Please test and give feedback

If you update/create the database by your own, it could go wrong and the plugin stops working ;-)

Thx, in advance
Olli

notenoughtech, 2022-02-17:
I'll live my life dangerously :)
I'll check it out thanks for the heads up

OllisGit, 2022-03-18:
Hi @notenoughtech,
currently the selected spools were stored in the local octoprint plugin-settings (see ```<op>/config.yml```
[image]
These information will be move into the database (feature "external db")...will take some time.

OllisGit, 2022-03-18:
fyi: New Version 1.7.0 is released and include the MQTT feature.

notenoughtech, 2022-03-18:
Awesome. I'll give it a go 😍

notenoughtech, 2022-04-01:
Hi, 
My apologies for the late reply. Other projects got in the way before I could jump on this with the feedback. 

It's nice to see the events share the details with MQTT, thank you. I have a couple of suggestions:

Spool deselected:
databaseId = null;   - would be cleaner to retain the ID of the active spool otherwise, I have to keep track of the spoolId assigned to the toolId

spool object - any chance to expose all details from spool added event? (brand, cost comments all shebang) 

edits to spools - no event is being fired. That would be ok, if either whole spool object is shared via MQTT on load, or perhaps drop the array of spools with manual refresh in the plugin page?

Thank you for the changes made so far.

notenoughtech, 2022-04-18:
I came up with a workaround for now (cloning the db file). Not the prettiest way but it works. If you want to see what I made with that - check out this: https://notenoughtech.com/3dprint/smart-filament-storage/

closed 2022-05-29 by github-actions[bot]