OllisGit/OctoPrint-SpoolManager #247: Spool not selected occasionally

RobertHolscher, 2022-02-16:
When printing to Octoprint from my Cura, the job arrives with the right color in my preview (in cura the color and the name are specified and equal to the names in Spoolmanager.
When I start to print jobs, sometimes Spoolmanager gives the message that there is not a spool selected and if I want to continue to print. The spool is selected and the force spoolselection is set in the parameters. I can not make it reapear consistantly, but it is irritating that the spool is not counted down.
Is this a know problem and what can i do to make the Spoolmanager listen to the Cura selected material.

OllisGit, 2022-02-17:
Hi @RobertHolscher,

maybe I am wrong but I think you mention these two points:

1. No spool-popup if directly started from cura
2. Spool/Material selection out of cura

The first point is discussed here:
- Community  https://community.octoprint.org/t/intercepting-start-print-api-call/41502
- Related ticket: https://github.com/OllisGit/OctoPrint-SpoolManager/issues/85 

Short: When started from CURA via API-Call there is no option to show a popup in the browser...maybe via a special gcode command

The second point is just not implemented. Currently there is no data-connection between the slicer-settings and the spool selection.

There exists a couple of request to this. Look for  https://github.com/OllisGit/OctoPrint-SpoolManager/labels/FEATURE%3A%20FILAMENT%20SELECTION

Answers this your question?

RobertHolscher, 2022-02-17:
Hi, 

Thanks for your response 
It is slightly different what I am experiencing. 
I send the job from cura to Octoprint without start or load print: just store the job to be printed later by manual instruction . Then when in Octoprint I select the job and select print, it should pop up with the question to print with the selected spool. However sometimes (irregular behavior) it says that no spool is selected and no record is kept of the used filament. There is no way to force the spool use and I have not found what is causing this. 
So I would fit it under maybe a bug in the spoolmanager . 

On the second item: indeed it would be nice if there is a way in cura to send the spoolID with the job data, maybe a cura  plug-in {proprietary info} is able to send this info which can be caught by the spoolmanager

Point 1 is currently the most important, the second point is nice to have

Kind regards

Robert

Verstuurd vanaf mijn iPhone

RobertHolscher, 2022-02-17:
> hmmm..two question
>
> what cura OP-settings do you use?
>
> Do you use the little print-symbol or click on the job and then click a second time to start the print?

Hi,

I only selected the web cam and confirm print job options

In Octoprint I print with the icon you pointed in the picture

Kind regards 
Robert

Verstuurd vanaf mijn iPhone

closed 2022-03-30 by github-actions[bot]