OllisGit/OctoPrint-SpoolManager #248: Plugin Helpers support

bwees, 2022-02-17:
Is there support for using helpers in the plugin? I would like to write an other plugin that supports changing the spool via the webcam feed and QR codes but there does not seem to be any way to change the spool other than through the REST api which should not be called from within plugins. Would it be possible to add functionality of selecting a spool via a helper and also retrieving database information from a help too? 

Thanks

bwees, 2022-02-17:
Would it also be possible to have this feature ported to FilamentManager?

OllisGit, 2022-03-18:
Hi @bwees,

it is on my list. Beside the Spool-Selection function, what other helpers do you expect?

fyi: In the latest release V1.7.0 I added couple of Events so that other plugins are informed about some spool-data.  See https://github.com/OllisGit/OctoPrint-SpoolManager#developer---section

bwees, 2022-04-18:
Keep issue open

closed 2022-05-29 by github-actions[bot]

bwees, 2022-10-11:
I would expect in addition to the selection the ability to get a json representation of the full database and also to receive the spool attributes back when I select a spool with the event.