OllisGit/OctoPrint-SpoolManager #253: Easier way of selecting filament

benwrth, 2022-03-17:
Hi there,
I am linking using this plugin to manage how much filament each print is using however I have found it to be a bit tedious to select which roll of filament to use.

What if there was a widget in which you could select filaments quickly and easily as so:
[image]

Or even just a dropdown would be fine for many.

As far as I'm aware, there is nothing like this, however if I've missed something please do let me know!

Thanks,
Ben.

OllisGit, 2022-03-18:
Hi @benwrth,

in one of the early versions I had a simple dropdown, but the the "community" wants more filter-options.
Now it looks like this:
[image]
[image]

Your screenshot looks interesting for single tool-head printer, right? And the printer button is only enabled after a file is selected, right?

...hmmmm...maybe a plugin settings: toggle between show "quick selection" or show "advance filter selection" view.

benwrth, 2022-03-18:
Hi @OllisGit,
Thanks for your response. I was not even aware of the current widget so that was a surprise!

> Your screenshot looks interesting for single tool-head printer, right?
Yes, thanks. I did forget to consider multiple head setups whle making this as I only really considered my own use case. Whoops!

> the printer button is only enabled after a file is selected, right?
This mockup was based off the File Manager widget so that was the icon that was there previously, maybe something which resembles selecting it would be more appropriate.

I think adding that toggle you mentioned in settings would be a good option as it gives people most choice and flexibility.

Thanks,
Ben.

closed 2022-04-28 by github-actions[bot]