OllisGit/OctoPrint-SpoolManager #266: Temperature offsets not working?

Sleepy81, 2022-05-04:
I have enabled nozzle and bed temperature offsets in the plugin settings, but I'm not sure its working.
I have a sliced gcode at 220C, and input -15C in the spool temp window, expecting a -15C offset to be applied, for a 205C print temp, but the prints till prints at 200C, and the offset window in Octopi - Temperature tab shows 0C offset. I also tried to put the actual temperature I wanted in the spool configuration (205C), but still the same result.
Am I using it wrong here, or is it not working properly? 
I have restarted octoprint, reselected the spool and all other things I can think of.

SirFlour, 2022-05-06:
hello, i have the same issue. the offset is not shown in octoprint (eg 170°C) but on the Prusa Display it is shown as 175°C. SO i guess it is working but showing it just wrong. For the normal printing of 215 the offset is included and correctly shown as 220°C. I also wondering if i am missing a Setting or if that could be a bug. I am happy to test anything :)

freezeghost, 2022-05-24:
I have similar issue after reboot or restart of octoprint I have to switch off and on again to working propertly.

closed 2022-07-04 by github-actions[bot]