OllisGit/OctoPrint-SpoolManager #270: Deleted spool in tool, now database will not show

Sleepy81, 2022-05-14:
Had used up a spool. -> Deleted the spool while it still was selected as the spool in tool0. -> Octoprint hung. -> Rebooted Octoprint.
Now no spools show in the spool manager. Ale are "gone"
However, if I download the DB as CSV, all spools are there. It also says "Database connection successful"

In octoprint log i have the following lines:
2022-05-14 10:35:41,475 - octoprint - ERROR - Exception on /plugin/SpoolManager/loadSpoolsByQuery [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/api/SpoolManagerAPI.py", line 914, in loadAllSpoolsByQuery
    for selectedSpool in self.loadSelectedSpools()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/api/SpoolManagerAPI.py", line 143, in loadSelectedSpools
    "Last selected Spool for Tool %d from plugin-settings not found in database. Maybe deleted in the meantime." % i)
NameError: name 'i' is not defined

Here is the csv file of the database:
"Spool Name","Color Name","Color Code [hex]","Vendor","Material","Serialnumber","Density [g/cm3]","Diameter [mm]","Diameter Tolerance[mm]","Flow rate compensation [%]","Temperature [C]","Bed Temperature [C]","Enclosure Temperature [C]","Offset Temperature [C]","Offset Bed Temperature [C]","Offset Enclosure Temperature [C]","Total weight [g]","Spool weight [g]","Used weight [g]","Total length [mm]","Used length [mm]","First use [dd.mm.yyyy hh:mm]","Last use [dd.mm.yyyy hh:mm]","Purchased from","Purchased on [dd.mm.yyyy]","Cost","Cost unit","Note"
"PrimaSelect White","white","#ffffff","PrimaCreator","PLA PRO","-","1.24","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","172.0","251463","57669","02.05.2022 18:36","02.05.2022 23:52","-","-","395.0","kr","Temp: 190-220C max @ 50mm\s speed  Bed: 0-60C "
"White","white","#ffffff","Creality","PLA","-","1.25","1.75","-","-","220","60","-","-","-","-","1000.0","220.0","876.0","332601","291359","02.05.2022 18:36","02.05.2022 23:52","-","-","299.0","kr","Temp: 205-225C Bed: No info "
"Ruby Red","red","#ff0000","PolySmart","PLA","-","1.24","1.75","0.03","-","220","60","-","-","-","-","750.0","161.0","654.0","251463","219275","02.05.2022 18:36","02.05.2022 23:52","https://polyalkemi.no/","-","199.0","kr","Temp: 200-220C Bed: 0-60C "
"Red","red","#ff0000","Radius","PLA","-","1.24","1.75","-","-","215","60","-","-","-","-","750.0","220.0","632.0","251463","211899","02.05.2022 18:36","02.05.2022 23:52","-","-","249.0","kr","Temp: 195-215C Bed: 35-60C "
"Purple","purple","#800080","eSun","PLA+","-","1.25","1.75","-","-","220","60","-","-","-","-","1000.0","270.0","0.0","332601","0","02.05.2022 18:36","02.05.2022 23:52","-","-","219.0","kr","Temp: 205-225C Bed: 60-80C "
"PrimaSelect Ocean Blue","blue","#0000ff","PrimaCreator","PLA Glossy","-","1.25","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","0.0","249451","0","02.05.2022 18:36","02.05.2022 23:52","-","-","289.0","kr","Temp: 210-230C Bed: 20-90C "
"PrimaSelect Jungle Green","green","#008000","PrimaCreator","PLA Glossy","-","1.25","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","4.0","249451","1330","02.05.2022 18:36","02.05.2022 23:52","-","-","289.0","kr","Temp: 210-230C Bed: 20-90C "
"PrimaValue","Green","#008000","PrimaCreator","PLA","-","1.03","1.75","0.05","-","210","60","-","-","-","-","1000.0","220.0","91.0","403642","36731","02.05.2022 18:36","02.05.2022 23:52","-","-","249.0","kr","Temp: 180-210C Bed: No Info "
"PrimaSelect Gold","Gold","#ffa500","PrimaCreator","PLA Metallic","-","1.25","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","294.0","249451","97785","02.05.2022 18:36","02.05.2022 23:52","-","-","299.0","kr","Temp: 210-230C Bed: 35-60C "
"Fire Engine Red","red","#ff0000","eSun","PLA+","-","1.25","1.75","-","-","-15","0","-","-","-","-","1000.0","270.0","507.5223415585952","332601","168773","02.05.2022 18:36","09.05.2022 01:25","-","-","219.0","kr","Temp: 205-225C Bed: 60-80C Stringing at 0.7mm retract. Trying 1.2mm "
"Chopstick Red","red","#ff0000","PrimaCreator","PLA Glossy","-","1.25","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","20.0","249451","6652","02.05.2022 18:36","02.05.2022 23:52","-","-","289.0","kr","Temp: 210-230C Bed: 20-90C "
"PrimaValue","Blue","#0000ff","PrimaCreator","PLA","-","1.03","1.75","0.05","-","210","60","-","-","-","-","1000.0","220.0","362.0","403642","146119","02.05.2022 18:36","02.05.2022 23:52","-","-","249.0","kr","Temp: 180-210C Bed: No Info "
"Black","black","#000000","eSun","PLA+","-","1.25","1.75","-","-","220","80","-","-","-","-","1000.0","270.0","336.0","332601","111754","02.05.2022 18:36","02.05.2022 23:52","-","-","219.0","kr","Temp: 205-225C Bed: 60-80C "
"PrimaSelect Black","black","#000000","PrimaCreator","PLA PRO","-","1.24","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","0.0","251463","0","02.05.2022 18:36","02.05.2022 23:52","-","-","395.0","kr","Temp: 190-220C max @ 50mm\s speed Bed: 0-60C "
"PrimaSelect Matt Black","black","#000000","PrimaCreator","PLA Matt","-","1.38","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","745.1","225952","224465","02.05.2022 18:36","08.05.2022 14:54","-","-","279.0","kr","Temp: 205-220C Bed: 40-60C "
"PrimaSelect Matt Black","black","#000000","PrimaCreator","PLA Matt","-","1.38","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","0.0","225952","0","02.05.2022 18:36","02.05.2022 23:52","-","-","279.0","kr","Temp: 205-220C Bed: 40-60C "
"PrimaSelect Matt Black","black","#000000","PrimaCreator","PLA Matt","-","1.38","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","741.5","225952","223382","02.05.2022 18:36","05.05.2022 23:42","-","-","279.0","kr","Temp: 205-220C Bed: 40-60C "
"PrimaSelect Matt Black","black","#000000","PrimaCreator","PLA Matt","-","1.38","1.75","0.05","-","220","60","-","-","-","-","750.0","220.0","0.0","225952","0","02.05.2022 18:36","02.05.2022 23:52","-","-","279.0","kr","Temp: 205-220C Bed: 40-60C "
"PrimaSelect Matt Black","black","#000000","PrimaCreator","PLA Matt","-","1.38","1.75","0.05","-","-","-","-","-","-","-","750.0","220.0","0.0","225952","0","02.05.2022 18:36","02.05.2022 23:52","-","-","279.0","kr","Temp: 205-220C Bed: 40-60C "

fhessel, 2022-05-14:
Just encountered the very same problem. The error message tells you what happened:

> Last selected Spool for Tool %d from plugin-settings not found in database. Maybe deleted in the meantime.

**As a workaround**, I stopped Octoprint, opened the config file (usually `/home/$user/.octoprint/config.yaml`) and searched for the `plugins` > `SpoolManager` > `selectedSpoolsDatabaseIds` key. I had a single ID there which pointed to the now-deleted spool. By changing it to an existing ID (for me decreasing it by 1 just worked), and after that starting Octoprint again, my spools were back and the plugin worked again.

Sleepy81, 2022-05-14:
Yeah, also manager to get it back. The problem was the spool that it referenced did not exists in the db, but seems like it was still looking for it. 

I exported the db to csv, made sure the csv was fine (it was), and the deleted the db. Then I reimported the csv and everything is back again.

closed 2022-06-25 by github-actions[bot]

LumaLux, 2023-06-14:
other workaround can be to export the CSV and the import the same, seems to work for me

krylovsk, 2024-07-16:
The issue is still there indeed. The workaround with updating config suggested by @Sleepy81 works.