OllisGit/OctoPrint-SpoolManager #273: Create from template generates AttributeError: 'NoneType' object has no attribute 'totalWeight'

chrisgilldc, 2022-05-16:
Good morning. I'm running into an error when creating a spool from a template. I only encounter this when creating from a template, creating a new spool straight-up works fine.

Workflow is to select 'Add Spool', then 'Copy from Template' and select a template.

Values get filled in, although for some reason the 'Auto Calculation' option is set to 'Combined Weight', despite the template being set to 'Filament Amount'. I also get a red box around 'Combined Weight - Initial Total'. The template contains both an initial weight (1000g) and empty spool weight (222g) so I'd expect this to be pre-filled with 1222g. I also can't fill it in because auto calculation is in combined weight mode.

I change Auto Calculation to 'Filament Amount', and can now fill in the initial total, and make up some 'remaining' number, like 1000. Interface shows proper calculations for Used and Remaining under Filament Amount.

However, when I press save, I get the following errors in the log:

```
2022-05-16 09:15:35,746 - octoprint.plugins.SpoolManager - WARNING - Save spool failed. Inital loading not possible, maybe already deleted.
2022-05-16 09:15:35,747 - octoprint - ERROR - Exception on /plugin/SpoolManager/saveSpool [PUT]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/api/SpoolManagerAPI.py", line 973, in saveSpool
    newDatabaseId = self._databaseManager.saveSpool(spoolModel, withReusedConnection=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 1068, in saveSpool
    totalWeight = spoolModel.totalWeight
AttributeError: 'NoneType' object has no attribute 'totalWeight'
```

Somehow this isn't creating the right object, despite the interface showing no errors.

Additionally, this quietly fails in the interface and the spool just doesn't appear. At first I played with filters to see if I had it filtered out by accident. Nope, it's an error. That would be helpful to surface.

PingPongLee, 2022-06-06:
Confirming that i too have this issue

codefaux, 2022-07-06:
Confirmed, still present. Looking at Issues page, seems length calculation isn't working reliably either? So much for switching, this plugin looked **stellar**.

codefaux, 2022-08-06:
I'm considering this plugin abandoned and suggesting it marked as such in the OctoPrint plugin repository. Shame, it really had serious potential.

closed 2022-09-17 by github-actions[bot]

broderickhyman, 2023-12-11:
This plugin is under new management at [dojohnso OctoPrint-SpoolManager](https://github.com/dojohnso/OctoPrint-SpoolManager) where I have fixed the template issue in the [1.7.3 release](https://github.com/dojohnso/OctoPrint-SpoolManager/releases/tag/1.7.3). The new management is now registered in the [official Octoprint plugins list](https://plugins.octoprint.org/plugins/SpoolManager/) so you can install like normal.
@chrisgilldc @PingPongLee @codefaux