OllisGit/OctoPrint-SpoolManager #278: Color change support

Nohus, 2022-06-17:
When printing with multiple spools (color changes during printing), it is not possible to select correct filament.
When starting a print with color changes, I would expect the plugin to ask for all the spools to be used in sequence.

Sembazuru, 2022-07-04:
Or, prompt for a new spool at the M600 gcode command.

mythmon, 2022-08-05:
I don't think this should auto close. This seems like a valuable feature to include.

Nohus, 2022-09-05:
Activity

Nohus, 2022-10-07:
Activity

Dexter0007, 2022-10-25:
When I changed spool in the octoprint while changing filament, the plug-in only subtracted the used amount of filament from the last spool selected.

Hottabych-, 2022-11-25:
When this can't solved with the MMU2 select Plugin (https://github.com/tkoecker/OctoPrint-Mmu2filamentselect), it would be interesting to support such changes with this plugin . (I haven't tested jet.)

TWIISTED-STUDIOS, 2022-12-05:
Is no one actively supporting this plugin anymore, this is an issue I have seen raised multiple times and all I see is the bot auto-mark for closure and never actually answers the question of why this doesn't work when its listed as a feature in the features list.

Hottabych-, 2023-01-04:
It's a pitty ...

dimo414, 2023-02-05:
.

Hottabych-, 2023-03-08:
reset, bot

dimo414, 2023-04-09:
.

Hottabych-, 2023-05-10:
Who has turned the clock?

dimo414, 2023-06-12:
.

dimo414, 2023-07-12:
.

Hottabych-, 2023-07-12:
I fear ... https://github.com/OllisGit/OctoPrint-SpoolManager/issues/333 and hope we get a chance with our wish: https://github.com/OctoPrint/plugins.octoprint.org/issues/1213

Hottabych-, 2023-08-13:
This is like riding a dead horse ... that's really a pity.

dimo414, 2023-09-14:
.

Hottabych-, 2023-10-15:
To be or not to be, that's the question...

Hottabych-, 2023-11-15:
Good things take time...but not so much.

TWIISTED-STUDIOS, 2023-11-20:
.

Dexter0007, 2023-12-27:
.

closed 2024-02-12 by github-actions[bot]