OllisGit/OctoPrint-SpoolManager #281: Addd New Spool

kdennis182, 2022-07-06:
Just installed this add-in and there is no where to add a spool, not on the side bar nor in the add-in settings screen. I can seemingly import a CSV file, but then there is no spools to choose from. I saw this issue in another thread but it did not help. Running the latest version of octoprint on chrome. I have also tried on Microsoft Edge and have the same issue.

codefaux, 2022-07-06:
I actually just installed the plugin for the first time as well.

Adding a spool is done via the Spools *tab* which is added by the plugin. Ironically, once there, I have added a template but can't seem to add a spool based on the template, so...lol..

kdennis182, 2022-07-09:
Do no even see a "SPOOLS" tab either on the side bar or under the wrench menu. Figured out how to add via a csv file in the wrench menu then choose a spool and edit the spool thru the side bar but cannot manually add a new spool other than with the IMPORT CSV.

kdennis182, 2022-07-09:
[image]

This image does not even exist as far as I can find.

codefaux, 2022-07-09:
It's not on either side bar. It's not under the Settings (wrench). It's a tab. Temperature, Control, GCode Viewer, Terminal, [...] Spools. It's there with things like the Klipper tab, Cancel Objects, Print Job History plugins. In the image you provided, near the top, you can see the other stock tabs. Its order won't be the same as yours, likely.
If you have a UI customizer installed, make sure it isn't disabling the tab.
There may be a drop-down arrow on the right end of the screen beside the tabs which shows tabs who don't fit.

kdennis182, 2022-07-10:
Well....There it is! TY codefaux.... Now I love this add--in.....

closed 2022-07-10 by kdennis182

codefaux, 2022-07-10:
No worries, it took me a bit to spot it as well. Also @kdennis182 -- be aware of #273 -- Template feature doesn't seem to be working.