OllisGit/OctoPrint-SpoolManager #285: Plugin triggers a "please reload" dialog after every job

dimo414, 2022-08-02:
I've been noticing for a while now that if I leave OctoPrint open for too long it ends up displaying a "Please Reload" overlay dialog even though nothing should have changed with the server. A few searches found similar reports like https://github.com/OctoPrint/OctoPrint/issues/2950 which suggest the issue is plugins writing to the OctoPrint config unnecessarily. After snapshotting my config for a little while I hit another dialog today and this plugin appears to be a culprit:

```
$ diff -u /tmp/octoprint-config-2022-07-31_05.53.57.yaml /tmp/octoprint-config-2022-08-02_05.59.26.yaml
--- /tmp/octoprint-config-2022-07-31_05.53.57.yaml      2022-07-31 05:53:57.356692508 +0000
+++ /tmp/octoprint-config-2022-08-02_05.59.26.yaml      2022-08-02 05:59:26.685099734 +0000
@@ -49,7 +49,7 @@
     hideInactiveSpoolsInSidebar: false
     selectedSpoolDatabaseId: 3
     selectedSpoolsDatabaseIds:
-    - 1
+    - 34
   _disabled:
   - discovery
   - filamentmanager
```

I'm guessing this spool ID value should be stored somewhere other than the OctoPrint config, since updating it seems to trigger this "please reload" overlay unnecessarily.

dimo414, 2022-09-02:
AFAIK this is still an issue.

dimo414, 2022-11-04:
@OllisGit I don't mind this bug sitting idle, I'm sure you have better things to do, but could you turn off the auto-closer? I don't know what else to do than ping this every month which is just obnoxious for everyone involved.

dimo414, 2022-12-06:
.

dimo414, 2023-01-08:
.

dimo414, 2023-02-08:
.

dimo414, 2023-03-13:
.

mdziekon, 2023-03-27:
I'm also affected by this issue, especially since I'm running it on a rather slow to "reload" OrangePi Zero 2, where the "reload" might result in the printing process getting interrupted for a brief moment (which might introduce artifacts in the print).

I've observed that it shows up:
- after browser's window longer inactivity (eg. phone's browser getting put into background for some time, but without closing the Octoprints tab, then going back to the browser to now see the "please reload octoprint" modal)
- after desktop browser's inactivity due to eg. the computer being suspended for night (after suspension, when I revisit the still open browser)

This leads me to belivieve that there must be some sort of mechanism of re-verification of the config loaded by the browser and as stored on the server. It's a bit weird that the "change originator" (eg. desktop browser tab) has to reload the config, even though the config was actually changed by that originator. This is probably something that should be investigated in Octoprint.

However, if possible, it would be great to keep the spool id selection outside of the main config file to prevent it being in need of reloading.

dimo414, 2023-04-26:
.

dimo414, 2023-05-28:
.

dimo414, 2023-06-28:
😞

dimo414, 2023-07-31:
😞

dimo414, 2023-09-01:
.

dimo414, 2023-10-02:
.

dimo414, 2023-11-03:
.

dimo414, 2024-01-05:
.

closed 2024-02-16 by github-actions[bot]