OllisGit/OctoPrint-SpoolManager #290: Spool not selecting

solo2424, 2022-09-10:
When I send a job to OctoPrint from Cura, the spool is not being selected automatically and I am not being prompted to select a spool. In the print details panel, it just says Required Filament unknown.

solo2424, 2022-10-15:
I since moved to filament manager but would like this issue to be addressed so I can move back to Spool manager. From my understanding, Spool Manager is the replacement to Filament Manager.

scurfin, 2022-10-26:
Same problem for me, OctoPrint 1.8.6 Python 3.10.6

m-klecka, 2022-11-11:
Also having this problem, octoprint OctoPrint 1.8.6, Python 3.8.15, Spool manager 1.7.0

TheFlamingShark, 2022-11-21:
Another one for the pile, same issue, OctoPrint 1.8.6, Python 3.8.15, Spool manager 1.7.0. 
also w/ UI customizer the pop-up saying that a spool isnt selected doesnt appear.

closed 2023-01-01 by github-actions[bot]