OllisGit/OctoPrint-SpoolManager #292: Required Filament unknown for new files added via watched folder

MrSco, 2022-10-06:
reboot/restart octoprint required for 'Required Filament unknown' message to be resolved on new print files added via watched folder. after reboot, required filament is calculated and clicking print doesn't give warning that 'required filament unknown'.

PowerShellCrack, 2022-10-26:
I am having the same issue except even after reboot I still get 'required filament unknown' no matter what I do. I believe this is coming from cost estimator plugin though. But both the cost estimator and print job history are selected to use spool manager.  I did have the older plugin filament manager installed and if I select that it would work just fine, but I have fully migrated to this one because if its detailed settings. 
[image]

[image]

[image]

[image]

Printer: Anet A6
OctoPrint: 1.8.6
Python: 3.7.3
OctoPi: 0.18.0
Cost Estimator plugin: 3.5.0
Spool Manager plugin: 1.7.0
Print Job History plugin: 1.17.0

PowerShellCrack, 2022-10-26:
I found that if I uninstall the PrintGeniusPlugin, this fixes the issue.

closed 2022-12-07 by github-actions[bot]