OllisGit/OctoPrint-SpoolManager #294: Using spool template does not add to database or saves after use

PowerShellCrack, 2022-10-26:
If I attempt to add a new spool using a template; it does not save it; no matter what. However, if I create the spool manually; it does save it...but temporarily? Meaning I loaded a spool details, select the tool, and select for printing and clicked **save**. It shows up in spool list. But then I cancel a print and then go to select it from the list again, it no longer exists. 

[image]

[image]

[image]

Printer: Anet A6
OctoPrint: 1.8.6
Python: 3.7.3
OctoPi: 0.18.0
Spool Manager plugin: 1.7.0
Cost Estimator plugin: 3.5.0
Print Job History plugin: 1.17.0

PowerShellCrack, 2022-10-26:
i did notice that the count keeps increasing each time I add it, but I never see it in list, even after clicking reload.

PowerShellCrack, 2022-10-26:
Update, after reloading entire website, they show up. might be a refresh issue.

Hoaas, 2022-10-27:
Installed SpoolManager for the first time today. Only manage to add 1 spool. Reloading or restarting OctoPrint does not help. After ReCreating the database I could again only add one spool.

I had marked this spool as template, and tried to add new spools by using Add Spool > Copy from template...

When instead manually filling in, or clicking the added spool and using the Copy option I still manage to add spools, making me think it's an issue with Copy from template.

AlexVercammen, 2022-11-01:
Same issue. "Copy" or "Add Spool" w/ or w/o "Copy Template", the spool won't show until restarting OctoPrint

tmcgee123, 2022-11-14:
Same issue here, hoping it gets fixed soon as it makes me not able to use this plugin. Restarting doesn't always work for me, have to reboot the whole pi

tmcgee123, 2022-11-19:
Alright everyone. I had some extra time to dig into this and it looks like anything to do with copying from a template is currently broken. Do not use Copy at all. I'm going to add information here about what I found. Hopefully it makes a fix faster to release. If not I will see if I can get a PR out

**ROOT CAUSE**
**PUT plugin/SpoolManager/saveSpool** is returning a 500 due to `databaseId ` being a falsey value

Request payload

````
{
  "selectedForTool": "",
  "isFilteredForSelection": false,
  "version": "",
  "isSpoolVisible": true,
  "hasNoData": false,
  "databaseId": "",
  "isTemplate": "",
  "isActive": true,
  "isInActive": false,
  "displayName": "test from template",
  "density": 1.27,
  "diameter": 1.75,
  "diameterTolerance": 0.05,
  "flowRateCompensation": null,
  "temperature": null,
  "bedTemperature": null,
  "enclosureTemperature": null,
  "offsetTemperature": null,
  "offsetBedTemperature": null,
  "offsetEnclosureTemperature": null,
  "colorName": "test",
  "color": "#000000",
  "totalWeight": 1000,
  "spoolWeight": 222,
  "remainingWeight": 120,
  "remainingPercentage": 12,
  "totalLength": 327364,
  "usedLength": 288080,
  "usedLengthPercentage": 88,
  "remainingLength": 39284,
  "remainingLengthPercentage": 12,
  "usedWeight": 880,
  "usedPercentage": 88,
  "code": null,
  "noteText": "\n",
  "noteDeltaFormat": {
    "ops": [
      {
        "insert": "\n"
      }
    ]
  },
  "noteHtml": "<p><br></p>",
  "firstUse": "",
  "lastUse": "",
  "firstUseKO": "",
  "lastUseKO": "",
  "purchasedOn": "",
  "purchasedFrom": null,
  "cost": 26,
  "costUnit": "$",
  "vendor": "Inland",
  "allVendors": [
    "",
    "Inland",
    "IC3D"
  ],
  "material": "PETG",
  "labels": [
    
  ],
  "allLabels": [
    
  ],
  "totalCombinedWeight": "1222",
  "remainingCombinedWeight": "342",
  "drivenScope": "filament",
  "drivenScopeOptions": [
    {
      "text": "Filament Amount",
      "value": "filament"
    },
    {
      "text": "Spool Weight",
      "value": "spool"
    },
    {
      "text": "Combined Weight",
      "value": "spool+filament"
    }
  ]
}
````

Although it threw an error on the server, the real problem is that databaseId is not being passed. Here is the stack trace in case proof is needed

````
2022-11-19 00:08:21,684 - octoprint.plugins.SpoolManager - INFO - API Save spool
2022-11-19 00:08:21,694 - octoprint.plugins.SpoolManager - INFO - Load spool for update with database id ''
2022-11-19 00:08:21,745 - octoprint.plugins.SpoolManager - WARNING - Save spool failed. Inital loading not possible, maybe already deleted.
2022-11-19 00:08:21,747 - octoprint - ERROR - Exception on /plugin/SpoolManager/saveSpool [PUT]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/api/SpoolManagerAPI.py", line 973, in saveSpool
    newDatabaseId = self._databaseManager.saveSpool(spoolModel, withReusedConnection=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/DatabaseManager.py", line 1068, in saveSpool
    totalWeight = spoolModel.totalWeight
AttributeError: 'NoneType' object has no attribute 'totalWeight'
````

AlexVercammen, 2022-11-25:
Could there be a problem with filtering? 
When I use the copy feature to duplicate a spool but with a different color, the new color is unchecked. 
When I add a new vendor, that vendor is unchecked in the filter list.

tmcgee123, 2022-11-25:
@AlexVercammen yes that is what I'm experiencing as well. I was going to make an issue for that, but I feel like the copy from template needs to be fixed first so I didn't bother.

Even when creating a spool without a template, the new spool will not show up sometimes if it is a different vendor/material/color that wasn't available before.

AlexVercammen, 2022-12-26:
It is possible to copy or add new spool. The filtering is the culprit. If you add a new vendor or a new color, even with the filter set to “All” these additions get unchecked by default. All you have to do is toggle off/on the “All” option or check “on”  the new entries.

 

 

Thanks, Alex.

snowake4me, 2023-01-15:
So glad to find this thread - as I _really_ couldn't figure out what I must be doing wrong.  I'd had the same experience as noted by some above, with the filters, while adding new spools.  However - I am unable to make templates work, at all:  Pretty sure I understand the concept / intent: save the redundant entry to consistent values, for example across a particular material type.  Every time I save something from a template - **it doesn't save.**  Clearing / updating all filters - even restarting OctoPrint or rebooting the pi seems to have any effect.

Regardless - _loving_ this plug in.  Learning more of its use, and interactions with OctoPrint, and even OctoDash (the latter being what led me to this plug-in).  Thanks, for **all** the great plug-ins!

dkaser, 2023-02-22:
This is still a problem. (posting for bot)

vzombie, 2023-04-01:
This is still an issue for me (post for bot to reopen)

closed 2023-04-12 by github-actions[bot]

CobraA1, 2023-04-16:
Problem is still happening, will try the workaround when I get a chance.

broderickhyman, 2023-12-11:
This plugin is under new management at [dojohnso OctoPrint-SpoolManager](https://github.com/dojohnso/OctoPrint-SpoolManager) where I have fixed the template issue in the [1.7.3 release](https://github.com/dojohnso/OctoPrint-SpoolManager/releases/tag/1.7.3). The new management is now registered in the [official Octoprint plugins list](https://plugins.octoprint.org/plugins/SpoolManager/) so you can install like normal.