OllisGit/OctoPrint-SpoolManager #304: Spool loaded not being recognized when print is started.

theboz1419, 2022-12-13:
I have been getting a notification that there is no spool loaded for the print even though I have one set up.  If I remove spool and reload same spool, it may or may not recognize the spool and I have to repeat until it recognizes the spool loaded and I can print as usual.  Then after the print is finished and I want to print another part, again it may or may not recognize the spool is loaded and again will repeat above steps again to be able to print.

mikeamettler, 2022-12-14:
I have the same issue.
Octoprint 1.8.6
Spool Manager 1.7.0

Aliens82, 2022-12-19:
I have same issue. Octoprint 1.8.6 and SM 1.7.0 Got the message "missing metadat or spool-fields"

Electric3el, 2022-12-24:
After many tries I decided to ignore the Error Message and just print.
Result: the used filament was subtracted correctly! This means the message "no spool [...]" itself was wrong.
Im printing right now with an other spool, everything is fine

Exported CSV to see if there is any difference between the spools but can't find any relevant. However, here they are, maybe its useful.

"good" spool:
"Prusa PETG clear","lightgrey","#d3d3d3","Prusa","PETG","d70b432599","1.27","1.75","0.01","100","250","90","0","0","0","0","1032.0","207.0","150.12641574822297","337839","49145","14.10.2022 14:37","24.12.2022 12:01","Prusa3D","29.03.2022 00:00","29.99","€","Spool detail page | Prusament Schruaben = 14 gramm "

spool with error message:
"Jetblack 1","black","#000000","Prusa","PETG","ab167c0647","1.27","1.75","0.01","100","250","90","0","0","0","0","1039.0","207.0","771.9799765931555","340131","252723","14.10.2022 14:37","22.12.2022 14:33","Prusa3D","20.09.2022 00:00","28.49","€","Spool detail page | Prusament  Schrauben = 14 Gramm "

theboz1419, 2022-12-24:
@Electric3el Thanks, well thats a relief, I didn't want to start a print until spool loaded correctly.  That will save some time, as I would sit there clicking away intil it recognized the roll, correctly.

kaoszwerg, 2023-01-18:
I have the same issue, ignored message as well and i will check if spool will be updated after pring
no change during print
actually status stays at: Required Filament unknown

theboz1419, 2023-02-18:
Its still an issue and should be fixed

ficwiktor, 2023-02-19:
I found out that added spools can be used with OctoApp on Android. If I start print with browser, I still get "missing metadata".

closed 2023-04-02 by github-actions[bot]