OllisGit/OctoPrint-SpoolManager #305: Temperature Offset not working

MarshMalloW0007, 2022-12-14:
Hello Ollis,

I have tested the temperature offset but it does not work.

I have activated the options in the configuration of the plug-in.
[image]

I have indicated the offset in the spool configuration.
[image]

The gcode file has a nozzle temperature of 100° and a bed temperature of 10°.
I indicate the offsets 110° and 40° to obtain a final temperature of 210° and 50°.

During the preheating, the temperatures are well reached.
[image]

But when the printing starts, 
[image]
[image]
the temperature of the nozzle goes down to that of the gcode: 100°. Not that of the bed.

The printing is thus impossible.

MarshMalloW0007, 2022-12-17:
I did several tests and here is what I found out:
- OctoPrint does not support manual offers greater than 50.
- The bug that I found does not concern this plug-in but directly OctoPrint.

Sorry for the inconvenience.

closed 2022-12-17 by MarshMalloW0007

cinghialino, 2023-03-04:
@MarshMalloW0007 is there a way to update the code file temperatures with the ones set for the filament spool upon selection?