OllisGit/OctoPrint-SpoolManager #311: Cannot install on OctoPrint 1.8.6 and Python 3.9.16

nodu, 2023-02-14:
OctoPrint 1.8.6
Python 3.9.16

```
2023-02-14 04:32:24,253 ! error in OctoPrint-SpoolManager setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
2023-02-14 04:32:24,258 ! pillow >=6.2.0<7.0.0
2023-02-14 04:32:24,261 ! ~~~~~~~^
```

dkaser, 2023-02-23:
What version of pip do you have installed? I can install just fine without any changes using:

Python 3.10.6
Pip 22.3.1
OctoPrint 1.8.6

ronzelver, 2023-03-01:
Same here, I cannot install Spool Manager. I have the following versions:

Python 3.9.16
Pip 22.3.1
Octoprint 1.8.6

sammcj, 2023-03-03:
Confirming it's a problem for me as well - it seems it doesn't work with the latest OctoPrint (1.8.6) and Python 3.11.2.

After successfully(!?) installing it doesn't show up as installed.

ronzelver, 2023-04-04:
Why isn't there any response?

IchSchreiGleich, 2023-04-05:
Same here. Installed setuptools 65.7.0 (any <66+ works, I think) and the plugin installs. But I cant create any spool, some gui elements are missing.

ronzelver, 2023-04-08:
> Same here. Installed setuptools 65.7.0 (any <66+ should work) and the plugin installs.
> 
> ```
> python -m pip install pip==22.3.1
> python -m pip install setuptools==65.7.0
> python -m pip install wheel==0.38.4
> ```

You mean that the plugin is visible and functional? I've installed the same version you mentioned, but I still have the same result i.e. install completes but the plugin is not there after restarting Octoprint.

IchSchreiGleich, 2023-04-08:
> You mean that the plugin is visible and functional? I've installed the same version you mentioned, but I still have the same result i.e. install completes but the plugin is not there after restarting Octoprint.

Yes, it is fully functional. My mistake was that I was looking in the wrong place, it is configurable in OctoPrint Settings > Plugin > Spool Manager. But I was wondering how to add a spool. It's not very intuitive as the menu item is overhanging in my installation (too many plugins maybe) and is next to the virualization of temperatures, terminal commands, etc.

closed 2023-05-20 by github-actions[bot]