OllisGit/OctoPrint-SpoolManager #322: Dialog complained about no spool selected, even though one was and the plugin functioned otherwise as normal.

charredUtensil, 2023-03-20:
A print finished as normal. I went to do a print the next day and was confronted with this dialog:

[image]

As you can see, the spool "Mint" is still selected below. I didn't change the spool, any settings on the spool, or even turn off the printer. The only thing that's changed from the successful print yesterday is that I uploaded this new gcode file. I tried deselecting and reselecting the spool, and that appears to function normally. The server log also seems to think it was successful:
```
2023-03-16 17:43:47,398 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_deselected' with payload '{'toolId': 0, 'databaseId': None}' to event-bus
2023-03-16 17:44:02,385 - octoprint.plugins.SpoolManager - INFO - Store selected spool Mint for tool 0 in settings.
2023-03-16 17:44:02,385 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 3, 'spoolName': 'Mint', 'material': 'PLA', 'colorName': 'mint', 'remainingWeight': 737.5}' to event-bus
2023-03-16 17:44:05,161 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 3, 'spoolName': 'Mint', 'material': 'PLA', 'colorName': 'mint', 'remainingWeight': 737.5}' to event-bus
2023-03-16 17:44:05,496 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 3, 'spoolName': 'Mint', 'material': 'PLA', 'colorName': 'mint', 'remainingWeight': 737.5}' to event-bus
2023-03-16 17:44:06,053 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 3, 'spoolName': 'Mint', 'material': 'PLA', 'colorName': 'mint', 'remainingWeight': 737.5}' to event-bus
2023-03-16 17:44:06,295 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 3, 'spoolName': 'Mint', 'material': 'PLA', 'colorName': 'mint', 'remainingWeight': 737.5}' to event-bus
2023-03-16 17:44:21,530 - octoprint.filemanager.analysis - INFO - Analysis of entry local:slug_200.gcode finished, needed 524.69s
2023-03-16 17:44:51,832 - octoprint.plugins.SpoolManager - INFO - Send Event 'plugin_spoolmanager_spool_selected' with payload '{'toolId': 0, 'databaseId': 3, 'spoolName': 'Mint', 'material': 'PLA', 'colorName': 'mint', 'remainingWeight': 737.5}' to event-bus
```
There are no errors in the JS console either. In the "spools" tab, all the settings still appear to be present and correct:
[image]

**Did the same happen when all other 3rd party plugins are disabled?**
The only other plugin I had installed was Consolidated Tabs. Disabling it did not help. Everything else is from a stock OctoPi image.

**Log file**
```
2023-03-16 17:35:25,346 - octoprint.startup - INFO - ******************************************************************************
2023-03-16 17:35:25,350 - octoprint.startup - INFO - Starting OctoPrint 1.8.6
2023-03-16 17:35:25,352 - octoprint.startup - INFO - ******************************************************************************
2023-03-16 17:35:25,819 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2023-03-16 17:35:25,820 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2023-03-16 17:35:25,820 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2023-03-16 17:35:26,362 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2023-03-16 17:35:26,380 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2023-03-16 17:35:30,265 - octoprint.plugin.core - INFO - Plugin Filament Manager (1.9.1) is disabled.
2023-03-16 17:35:30,462 - octoprint.startup - INFO - Added settings overlay from plugin consolidatedtabs
2023-03-16 17:35:30,463 - octoprint.plugin.core - INFO - Found 22 plugin(s) providing 21 mixin implementations, 44 hook handlers
2023-03-16 17:35:30,669 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2023-03-16 17:35:30,817 - octoprint.server - INFO - Intermediary server started
2023-03-16 17:35:30,860 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2023-03-16 17:35:30,916 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2023-03-16 17:35:30,925 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2023-03-16 17:35:30,926 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2023-03-16 17:35:30,926 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2023-03-16 17:35:30,927 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2023-03-16 17:35:30,928 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2023-03-16 17:35:30,929 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2023-03-16 17:35:30,930 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')")
2023-03-16 17:35:30,932 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2023-03-16 17:35:30,936 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2023-03-16 17:35:30,939 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2023-03-16 17:35:30,942 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2023-03-16 17:35:30,945 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_LIST (needs: "Need(method='role', value='plugin_pluginmanager_list')")
2023-03-16 17:35:30,948 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2023-03-16 17:35:30,950 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
2023-03-16 17:35:30,953 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2023-03-16 17:35:30,957 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2023-03-16 17:35:30,960 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2023-03-16 17:35:32,966 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2023-03-16 17:35:34,836 - octoprint.util.pip - INFO - Version of pip is 22.3
[... cut ...]
I also tried restarting the entire computer multiple times, but this didn't resolve it. However, once the print finished, the total remaining filament updated correctly - and then I was able to print several more things without this reproducing.

**Screenshots**
Inline

**Additional context**
I recently switched from FilamentManager to SpoolManager but this is the first time I'm seeing this issue. Again, I changed _absolutely nothing_ from yesterday when this was working fine. My hunch is that maybe this doesn't work if the backend isn't done analyzing the file, and the error message is just confusing?

SirFlour, 2023-03-28:
Same message for me :_(

charredUtensil, 2023-03-28:
So after more testing, I believe this is a misleading error message, and the cause is actually SpoolManager failing to compute filament usage in the gcode file. The reason I think that is I have one gcode file that _always_ generates the error: It's for leveling the print bed. It doesn't doesn't extrude any filament or heat up the nozzle.

I'm not very familiar with this codebase, but the error seems to be decided here:
https://github.com/OllisGit/OctoPrint-SpoolManager/blob/dea8d64c1849c970f3616e158260c4c6fef5a4b7/octoprint_SpoolManager/api/SpoolManagerAPI.py#L378

SirFlour, 2023-03-30:
i have the same effect. some gcode work and newest one not. i changed to the latest prusa-slicer version, maybe something changed in the gcode. Or maybe some settings in the slicer?

charredUtensil, 2023-04-01:
I can also consistently reproduce this bug by uploading a different gcode file with the same name (overwrite) and then immediately printing it. After a few minutes and a few failed prints, FilamentManager eventually stops complaining and behaves normally.

rubienr, 2023-04-30:
I can reproduce it as follows:
1. slice and upload file to OctoPrint (preferably a larger one)
1. immediately try to start the print **before OctoPrint has finished the file analysis**
1. spool manager will say:
    ````
    ATTENTION: Needed filament could not calculated (missing metadata or spool-fields)

    There are no spools selected for the following tools despite them being used (maybe) by this print:
    - Tool 0
    - Tool 1

    Do you want to start the print without selected spools?
    ````
In contrast if i wait until the fie has been analyzed the pre-selected spool is chosen as expected.

Since i have larger files to print and OP is running on a BananaPi, it may take ages until the file is processed. Thus i wish i could disable "Settings -> Features -> Features -> Analyze gcode for time and model size estimates -> Never". But disabling this feature with `OctoPrint: 1.8.7` + `Spool Manager: 1.7.0` will never allow me to select any spool (as described above).

**Suggestion:** If the user starts a print of a file that has not been analyzed so far, still allow the user to start the print with the preselected spool.

**Workaround:** Leave file analysis enabled and wait until OP reports the `Approx. Total Print Time: ` as "xx minutes" instead of "-". This is an indicator that the file processing has finished.

closed 2023-06-11 by github-actions[bot]