OllisGit/OctoPrint-SpoolManager #325: Feature Request: QR Code Print Button

ScottGibb, 2023-04-18:
### Feature Summary

When using the plugin, I found it to be good at tracking my spools. I especially like the QR code that links to each spool on the database and the ability to select one by simply scanning it. I have possibly quite a niche but interesting request in that, I currently own the Dymo LabelWriter 450 printer, which I hope to make accessible to my entire network as a networked printer. In the GUI for the plugin, when setting up the spool, it would be ideal if there were a print button next to the QR code which would allow me to connect to print off the QR code when I set up a new entry in the database.  So the workflow would be as follows, as for the code for networking the LabelWriter I have linked it below:

New Filament Setup in Plugin-> Press Print QR Code -> Stick on to Spool

## Useful Links

- [Raspberry Pi Server Dymo LabelWriter](https://www.hmazter.com/2013/05/raspberry-pi-printer-server-for-labelwriter/)
- [Dymo LabelWriter Ask Ubuntu](https://askubuntu.com/questions/500242/i-need-to-install-dymo-software-for-labelwriter-450-but-i-cant-any-help) 
- [Dymo LabelWriter 450](https://www.dymo.com/label-makers-printers/labelwriter-label-printers/dymo-labelwriter-450-direct-thermal-label-printer/SP_95488.html)

## Acceptance Criteria

- Button added to UI to allow the printing of QR Codes through a networked, or USB Printer

MrRexRuthless, 2023-04-23:
Love this idea.  Second this

ScottGibb, 2023-05-10:
For anyone that stumbles across this feature request, I've managed to get the Dymo LabelWriter 450 setup using Docker and a Pi zero, the Dockerfile and setup process is outlined in the following repository with links to the tutorials that helped me piece it all together.

# Raspberry Pi CUPS Dymo 450 LabelWriter
https://github.com/ScottGibb/Cups_Dymo-450

ScottGibb, 2023-06-10:
Adding a comment, so this issue will remain open.

closed 2023-07-23 by github-actions[bot]