OllisGit/OctoPrint-SpoolManager #329: BUG: When Scanning QR Code of different spools, only spool 1 is selected for printing

ScottGibb, 2023-05-10:
# Issue

When using the plugin, I found that no matter which spool QR Code I scanned, the plugin would always select spool 1 in the database for printing. I'm running an iPhone 13 with the built-in QR Code Scanner, opening the webpage in Safari. When I scanned spool 2 in the database through the UI, the following link was created:

_http://wanhaoprint.local/login/?redirect=%2F%3F&permissions=STATUS%2CSETTINGS_READ#tab_plugin_SpoolManager-spoolId2_

It would seem that the URL is being created correctly, however possibly not being handled on the other side correctly.

A part of this, awesome plugin!!! Really enjoy using it, and looking forward to getting it integrated with my Label Maker!! (#325) 

# Steps to Repeat

- Use an iPhone 13 QR Code Scanner
- Scan the QR Code on a spool that isn't the first in the database.
- Plugin selects spool 1 for printing

Ticket #185 says this was fixed in V1.5, in my system i'm running V1.7.

# Hardware/Software Setup

- OctoPrint-SpoolManager v1.7
- Octoprint Version 1.8.7
- Raspberry Pi 3

normen, 2023-07-26:
Meep

ScottGibb, 2023-08-27:
moop

mdziekon, 2023-08-28:
This issue most likely has been already fixed in my fork of the plugin, see:
https://github.com/mdziekon/OctoPrint-SpoolManager/issues/37

ScottGibb, 2023-09-01:
@mdziekon Awesome, il give it a shot this weekend hopefully. Is there anyway to keep my database for spools intact when uninstalling this plugin and then installing yours?

mdziekon, 2023-09-01:
@ScottGibb when using either URL or ZIP install, my fork should simply replace your, without affecting the DB. To be safe, make sure to perform a DB backup though.

ScottGibb, 2023-09-11:
@mdziekon Sorry for taking so long to test it, but I've now installed your plugin, and tried both 1.8 and 2.0 and both do not work, The same issue as before keeps happening. When you scan the QR code it creates the correct URL as before, however, it continues to select the first spool in the database.

normen, 2023-10-12:
huargh

ScottGibb, 2023-11-12:
beep

normen, 2023-12-18:
I switched to this branch, seems to have all the fixes we need: https://github.com/jkotowicz1/OctoPrint-SpoolManager

Its settings are compatible, you just need to uninstall this version and install that version above using the .tar.gz rile

ScottGibb, 2023-12-19:
@normen Cheers for the link, sadly that repo also has the same issue as this. I'm gonna copy and paste the issue into that repo and see what happens

normen, 2023-12-19:
> @normen Cheers for the link, sadly that repo also has the same issue as this. I'm gonna copy and paste the issue into that repo and see what happens

Sorry, I gave the wrong link, this is the version I have installed (so many forks!)
https://github.com/mdziekon/OctoPrint-SpoolManager

normen, 2023-12-19:
Omg, could it be this one? It has the links to the mdziedkon one so it looks like that in the settings for me too.. 😕

https://github.com/jkotowicz1/OctoPrint-SpoolManager

ScottGibb, 2023-12-19:
@normen YESSSS!!!! That is the one that works, so so many forks hahah. For anyone that comes across this issue: This the fork you want to use:
https://github.com/jkotowicz1/OctoPrint-SpoolManager

ScottGibb, 2023-12-20:
@normen, the other repository that @mdziekon made also has the bug fix with more features. If your interested, here is the issue discussion, where mdizekon helps me fix the qr bug, needed to checkout the prerelease code.
https://github.com/mdziekon/OctoPrint-SpoolManager/issues/52

closed 2024-02-02 by github-actions[bot]