OllisGit/OctoPrint-SpoolManager #331: error in OctoPrint-SpoolManager setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)

reini-1, 2023-06-09:
I cannot install/update the plugin, because it shows the error below.
Is there a workaround/fix available?
Is the plugin still in development or is it deprecated/abandoned?

Thanks in advance.

octoprint@octoprint4:~ $ pip install https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip
  Downloading https://github.com/OllisGit/OctoPrint-SpoolManager/releases/latest/download/master.zip (670 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 670.1/670.1 kB 5.7 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [4 lines of output]
      error in OctoPrint-SpoolManager setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
          pillow >=6.2.0<7.0.0
                 ~~~~~~~^
      Found packages: {'octoprint_SpoolManager.common', 'octoprint_SpoolManager.api', 'octoprint_SpoolManager.models', 'octoprint_SpoolManager'}
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

octoprint@octoprint4:~ $ python --version
Python 3.9.2

octoprint@octoprint4:~ $ pip --version
pip 23.1.2 from /opt/octoprint/lib/python3.9/site-packages/pip (python 3.9)

octoprint@octoprint4:~ $ uname -a
Linux octoprint4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

octoprint@octoprint4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

octoprint@octoprint4:~ $ octoprint --version
octoprint, version 1.9.0

WoosterInitiative, 2023-06-28:
In *setup.py*, `"pillow>=6.2.0<7.0.0"` needs a comma, like `"pillow>=6.2.0,<7.0.0"`.

In some environments setuptools would fallback to the setup.py method when it couldn't build the wheel because of this, but on others (notably when I'd use paukstelis' *octoprint_install* or *octoprint_deploy* scripts) it would just fail. Haven't been figure out what's different but it doesn't matter, this should be fixed.

I'll do a PR for this.

WoosterInitiative, 2023-06-28:
Nevermind, there's already an ignored PR for the exact same change. #327

reini-1, 2023-08-02:
@OllisGit Is this project dead?

Sporksan, 2023-08-02:
hey @reini-1, I had the same issue after having to restore my instance from a crash.  To fix this i followed the instructions on #327 
In your Octo print plugin manager use the following URL to grab a bugfix version of the spool manager plugin.  you may have to scroll down in the "Get more plugins" dialog to be able to manually enter the URL.
https://github.com/HTXF-g/OctoPrint-SpoolManager/archive/refs/heads/bugfix.zip

closed 2023-09-14 by github-actions[bot]