OllisGit/OctoPrint-SpoolManager #333: Is this dead?

WoosterInitiative, 2023-06-22:
It doesn't look like this has been touched since March 2022, more than a year ago. It doesn't seem to want to install on Ubuntu jammy, at least for me, and I'm not sure whether it's worth trying to figure it out

darthbeluga, 2023-06-22:
I am trying to do the same. I was just wondering that as well.

CobraA1, 2023-07-01:
Looks like OllisGit has been pretty inactive the past year or so, only 9 commits (none to this particular project) in the last year.

It still works, I've been using it. But it looks like a few bugs are unlikely to be resolved.

wizardkitty, 2023-07-05:
OllisGit Maintains the only Two filament mangers for octoprint. both are now effectively abandoned with basically the same issue. If anyone could fix it that would be welcome.

mdziekon, 2023-07-06:
@wizardkitty take a look at my fork of this plugin: https://github.com/mdziekon/OctoPrint-SpoolManager

Recently I've been fixing some performance issues I had with my OctoPrint instance, did some fixes to this plugin as well, but since it's effectively dead (as in, development is dead, not the plugin itself which works rather fine), I've been pushing every change to my fork only. It's not much, I won't promise to fix every issue there is with this plugin, but I guess it's the closest thing to maintenance you can get at this moment.

wizardkitty, 2023-07-07:
@mdziekon Well. The same person also maintained the filament manager plugin and it seems both have run into the same issue of installation and its only going to get worse with time as other linux distros get updated. Ill probably look into reporting both of these plugins as abandoned in a couple weeks if no response is given to this issue or the one in filament manager asking if its been abandoned.

Kehet, 2023-07-27:
@OllisGit seems to be alive 🎉

https://github.com/OctoPrint/plugins.octoprint.org/issues/1213#issuecomment-1647442098

dimo414, 2023-08-27:
@OllisGit says they're [still here](https://github.com/OctoPrint/plugins.octoprint.org/issues/1213#issuecomment-1647442098). 

OllisGit, if you see this it would be _amazing_ if you could turn off the auto-close GitHub action on your repos.  I understand you can't respond to everything immediately but in the meantime the auto-closer is causing harm.

dimo414, 2023-09-27:
.

flaviut, 2023-11-05:
.

challgren, 2023-12-07:
.

drohhyn, 2024-01-08:
The addon was taken over, have a look here: https://github.com/dojohnso/OctoPrint-SpoolManager
There is already a new version in the plugins repo.

ThomDietrich, 2024-01-12:
Any recommendation which fork to go with?
Seems like there are three considerably active forks:
- https://github.com/dojohnso/OctoPrint-SpoolManager by @dojohnso
- https://github.com/mdziekon/OctoPrint-SpoolManager by @mdziekon didn't see updates in 6 months
- https://github.com/jkotowicz1/OctoPrint-SpoolManager by @jkotowicz1, a fork from @mdziekon 's latest

Did you consider joining forces? You guys could form a GitHub organization and manage some of the abandoned repos under that :)

Kehet, 2024-01-12:
@ThomDietrich [Octoprints plugin repo](https://plugins.octoprint.org/plugins/SpoolManager/) seems to point to https://github.com/dojohnso/OctoPrint-SpoolManager, so I think that would be considered official fork now

ThomDietrich, 2024-02-12:
Still relevant. Let's keep this one active for updates on forks etc

drohhyn, 2024-02-12:
> Still relevant. Let's keep this one active for updates on forks etc

No, it's not.
As already mentioned earlier, the plugin is still in the official repo under https://plugins.octoprint.org/plugins/SpoolManager/ AND it points to this github repo: https://github.com/dojohnso/OctoPrint-SpoolManager. Both links are in this discussion within the last 5 comments.

Updates on forks can be found in the forks, there is no more reason to have it here. The most relevant fork is the one from dojohnso because this repo builds the official plugins.

ThomDietrich, 2024-02-12:
This repo does not seem properly maintained.
There is no harm in keeping this "information thread" open for future news/discussions. You don't have to care about it and may unsubscript from it

drohhyn, 2024-02-12:
> This repo does not seem properly maintained.
I am not associated with the plugins maintainer in any way (but a user myself), but: last commit 4 days ago, several features branches + release + "official repo" as listed on octoprint.org.
> There is no harm in keeping this "information thread" open for future news/discussions.
Correct, but the question was "is this dead"? Answer: yes AND there is ONE relevant fork. This is the important info for all those coming from old plugin installations, thus this thread is answered and should be closed (which would have happened automatically). You can still find and build your preferred fork, but the official one is still https://github.com/dojohnso/OctoPrint-SpoolManager
> You don't have to care about it and may unsubscript from it
Done this already, because the answer is given several times.

Ajimaru, 2024-02-27:
That worked for me on OctoPrint 1.9.3 Python 3.10.12 [https://github.com/OllisGit/OctoPrint-FilamentManager/issues/119#issuecomment-1624481946](url)

closed 2024-04-08 by github-actions[bot]

ThomDietrich, 2024-09-06:
https://github.com/OctoPrint/plugins.octoprint.org/issues/1319